kareltucek/firmware #15: Mouse recording for the macro recorder?

kareltucek, 2019-01-26:
The idea is that mouse position is reset to (0,0), then activation times of mouse are recorded and later replayed, so the mouse movement happens in "real time" and is not affected by device driver's acceleration/deceleration rules.

- How should simultaneous activations of different mouse movements be handled?

kareltucek, 2019-02-01:
- how should dynamic macroes be handled? (We don't have such macroes yet (at least not any useful ones), but once we get them, it would be nice to be able to use them in the macro recorder while retaining their dynamic nature.)

IzK666, 2020-06-18:
Although not the same issue, it's related.
Can you include mouse buttons into macro recorder?

kareltucek, 2020-06-21:
Theoretically yes. Practically it is not a trivial task and I do not plan extending the macro recorder in the near future. 

Do you have any specific usecases in mind? In my experience, mouse clicking is usually tied to mouse movement anyway, and therefore it is usually feasible to use an actual mouse in one hand and do the clicking, while activating (keyboard-only) macros by the other hand...

IzK666, 2020-06-30:
I was playing the other day with a browser game (it wasn't an idle game, but quite) and I needed to click a lot.
I can do it with a regular macro.