kareltucek/firmware #24: [idea] virtal mod keys composed of two physical keys

orthoceros, 2019-04-22:
Here is an (untested) idea for a virtual mod key composed of two physical keys, triggered by a 2-finger gesture. The rationale is to get an easy, reliable and quick activation gesture for secondary actions on alphabetic keys. It should almost never interfere with normal typing and gives us arbitrary mod keys without having to move fingers far from the base row.

- trigger: press the 2-finger-mod (e.g. c+v) simultaneously and support an arbitrary press order. I.e., we either receive pressC->pressV or pressV->pressC, without any other interrupting key event.
- then, resolve as secondary if:
  - at least one trigger key is not yet released AND
	- either a third keyPress is received 
	- OR at least one trigger key is pressed longer than 500ms. This delay is important to cancel a secondary action without receiving unwanted typing actions; in this case, the user just must not release the two-key-mod too early.
- if resolved to a secondary action: First, consume the two postponed trigger keys. Then, either implement the action directly in the macro (faster) or hold a keymap layer until both trigger keys are released (more flexible, as it allows implementing all actions for possible third keys like copy/paste actions or other macros on GUI level in Agent).
- normal typing in all other cases, i.e. goTo `$postponeKeys holdKey thisKey` (postponeKeys should be used everywhere to assert the correct processing order in concurrent smashing cases)

Example gestures:
 - An easy time line example would be to `hold(c+v), tap(j), release(c+v) => play action on MAK.mod.j`. But I would also like to enable one of the trigger keys as third keyPress, i.e. allow a dual role as actionKey, e.g.:
 - Hit c+v in arbitrary order (semantics of one logical mod key composed of two physical keys), releaseV, pressV, release both in arbitrary order => map to an action, e.g. LCtrl+V. So, between releaseC at the end and pressC at the beginning (assuming it came after pressV), we have at least two events (a release and a press). This should be enough to disambiguate it from normal typing of cvv in almost all cases.
 
[Ideally the following additional failsave logic is not necessary at all, but it will require testing:] On top of this base idea, it might be possible to rescue all `cv.` and `vc.` tripletts for normal typing, if the `.` means a third key that is not mapped to any defined action (and if both trigger keys are released faster than 500ms). This does not have to be an inefficient lookahead (changing the keymap for if branching), but instead possibly could be realized with a generic return macro on the target layer, for all keys that were previously assigned the none action. But this would not be trivial (undoing the consumePending of the two trigger keys in the correct order...) and hopefully totally unnecessary, thanks to the more elaborate trigger logic.

kareltucek, 2019-04-22:
This should give a reasonable approximation (just fill in the keyids):

    $ifNotShortcut c v goTo default
    $holdLayer secondaryLayer
    $ifNotPlaytime 500 ifNotInterrupted write cv
    $break
    $default:
    $tap c
 
Also, you can directly map the cvv variant as:

    $ifShortcut v v tapKey ...

But generally, I think that you should start thinking about messing with the firmware code directly ;-)

orthoceros, 2019-04-23:
I had played with `ifShortcut` before the mutex problem, but did not think of using `ifNotShortcut` to implement the trigger logic. Looks promising; thanks. I will try it and then report back here (in a week or so).
With respect to the source code: I had a glimpse at it, but it is not my language and I have no experience at embedded programming. To make myself useful on this level would take quite some time that I simply don't have in the foreseeable future...

kareltucek, 2019-07-14:
@orthoceros how did it work out? Did you try to use the "dual" modifier approach in your workflow? Was it useful or were there some fundamental issues (either with the implementation or with the idea itself)?

orthoceros, 2019-07-17:
Hi. I just updated from kt.16 to kt.21. (Sorry, I had no time earlier and am still low on time. But I started trying to replace my mouse with the UHK keyboard mouse a few weeks ago... ;) Anyway, I had a brief look at your current docu today: good write-up and cool new grammar; especially thanks for the comment rules! 

I tried to implement the "virtual-mod-key-by-a-pair-of-two-physical-keys" idea of this thread today, using your above suggestion of directly mapping the three-key combos with your current grammar:
```
//Macro for the c key: (Are full comment lines skipped correctly or do they require a $postponeKeys modifier?)
$ifGesture 090 089 final tapKey LC-c  //c->v ->c => copy
$ifGesture 090 090 final tapKey LC-v  //c->v ->v => paste
$holdKey c
//Macro for the v key:
$ifGesture 089 089 final tapKey LC-c  //v->c ->c => copy
$ifGesture 089 090 final tapKey LC-v  //v->c ->v => paste
$holdKey v
```
While the c+v trigger works, I unfortunately get `ocmment` again when typing fast/two-handed, instead of `comment` etc. The latter also happens with `$ifShortcut` and with `$postponeKeys ifShortcut`. Not sure why. (As this macro pair stands, it also yields `ccccccccccccv` or `vvvvvvvvvvvvc` when just hitting `cv` or `vc` and then releasing both keys. Probably, the release of the first key is only played after the 1000ms wait period of the `$ifGesture` in the macro of the the second key, causing OS key repetition.)
____

Two hours ago, I came up with the following more general idea for key `c` (and its symmetric version on `v`). I hoped that this macro could automatically/elegantly release any typed key in the correct typing order, after checking for defined 3-key-gestures:
```
$postponeKeys holdKey c //Enqueue pressC at the correct point in time, as if this key was not mapped to any macro at all! (And enqueue releaseC later, whenever the user releases this key.)
$resolveNextKeyEq 1 090 200 @1 @3  //for 200ms, look ahead for the trigger pair "pressC (at queue index 0) ->pressV (at queue index 1)"
$break  //remove the postponeKeys modifier to play the enqueued default (and, possibly, any other commands enqueued by other key presses in the meantime)
//Look-ahead for defined gestures:
$resolveNextKeyEq 2 089 400 @1 @3  //in a new 400ms time window, wait for the 3rd key (at queue index 2). Implement (c+v)->c => LCtrl+C in the next two lines.
	$consumePending 3 //consume c,v,c (Does it also consume associated release events in the queue, if they happened in the meantime? How are releases counted; is 3 always the correct number to consume?)
	$final tapKey LC-c
$resolveNextKeyEq 2 090 0 @1 @3  //Implement (c+v)->v => LCtrl+V. Reuse the above 400+0ms time window, i.e. just check the current queue for another defined gesture (3rd key ID).
	$consumePending 3 //consume c,v,v
	$final tapKey LC-v
//release postponeKeys to perform enqueued default actions in the correct order, exactly as they were typed (even if other keys interrupted).
```
However, I need help, as this idea seems to already fail at line 1: The `$postponeKeys holdKey c` does not seem to enqueue `c`, but executes `holdKey` immediately. (I see the `c` appear without any delay.) The same happens with `pressKey` instead of `holdKey`. I also tried `$postponeNext 3` and then `$call keyPressMacro`, but still get `c` immediately. Is there any way to _enqueue a macro command_ like `$pressKey`?

kareltucek, 2019-07-17:
> While the c+v trigger works, I unfortunately get ocmment again when typing fast/two-handed

I am very sorry about this - I've been writing small improvements and bug fixes ever since that release but did not publish them yet. I think that it should be fixed with 8.5.4.kt.22, which I am pushing right now.

> As this macro pair stands, it also yields ccccccccccccv or vvvvvvvvvvvvc when just hitting cv or vc and then releasing both keys. Probably, the release of the first key is only played after the 1000ms wait period of the $ifGesture in the macro of the the second key, causing OS key repetition.

Yep correct. The postponer should be rewritten so that it postpones modifiers only. I know about this problem, however solving it it would need another big cut into the code.

> Are full comment lines skipped correctly or do they require a $postponeKeys modifier

The same policy still applies, i.e., they need postponeKeys modifier. I agree that it is not an ideal state.

>  Is there any way to enqueue a macro command like $pressKey?

No. The postponing mechanism postpones the actual "hardware" presses in preprocessing phase and does not affect scancode sending at all. 

Anyway the first macro is totally correct. If anything malfunctions, the problem is with the firmware, not with the macro :-). 

> Does it also consume associated release events in the queue, if they happened in the meantime? 
> How are releases counted; is 3 always the correct number to consume?

All postponing queue operations relate to presses. If releases need handling, they are handled implicitly. I.e., yes and yes.

orthoceros, 2019-07-19:
Thank you for the update and your answers! I upgraded to kt.22 and tested the above `ifGesture` macros again:
 - writing fast: Comment comment comment comment comment command comment comment command comment command... seems to work now! :)
 - The problem with delayed release events seems to be solved, too (hitting just cv or vc now only yields cv or vc after 2s, i.e. no OS repetitions like cccccccccccv etc. any longer).
 - Copy/paste gestures seem to trigger reliably when entered on purpose.
 - holding a single key to trigger OS key repetition on purpose also works: cccccccccccccccccccccccc...

I have also performed some usability tests:
- The delay from `ifGesture` is quite noticeable, not only for cv and vc, but also for single c or single v or composites like Shift+C...all appear after a too long delay. The time between the pressC and pressV is always less than 200ms when thinking of the trigger c+v as a "virtual-mod-composed-of-two-keys": From a user perspective, the trigger gesture is to press them simultaneously. The third key in the gesture sequence should follow this trigger in a 800ms time window. So, I am thinking of something like `$ifGestureWaitMax 090 200 089 800 final tapKey LC-c`.
- Testing side effects: When I tried to write cvcvcvcvcvcv..., this only worked when actively waiting for the ifGesture delay; otherwise I triggered unintended gestures. However, after both physical keys of the virtual-mod-composed-of-two have already been released, it logically should no longer listen for any gestures. Then cvcvcvcv.... could then be entered at normal speed
- This is already the case when using `ifShortcut` instead of `ifGesture`. But the price here is that the trigger "pressC->pressV" can no longer be used to enter the (c+v)->c gesture, because this corresponds to the sequence (pressC->pressV)->releaseC->pressC and `ifShortcut` already breaks at the release of the first pressed key. Could `ifShortcut` get an option to keep listening until the release of the _second_ key in the sequence? I think this would nicely provide the intended logic of a virtual-mod-key-composed-of-two and solve above time window problems. (Logically, this virtual mod key should be treated as not released until _both_ of its comprising physical keys have been released...)
- Speaking of time windows, the user should get an easy "I changed my mind/cancel!" option after he pressed the c+v trigger. In the `ifShortcut` case, this could probably be implemented via `$ifPlaytime 1500 break`, i.e.: "do not release c+v for long enough and nothing will happen"; just like when holding a usual mod key for some time. However, a `ifShortcutWaitMax` version could still be required (to specify a short time window until the second key press and an `untilAllPendingKeysAreReleased` window for the third key)... At least I currently do not see any other way to get both, this cancel option and rescuing the OS default behavior ccccccccccccccccccc... when holding just a single key.

(another post that zooms out a bit from this detail level follows soon...please wait.)

orthoceros, 2019-07-19:
I have been thinking (longer than originally intended, but it was fun...;) about a possibility to unify and to generalize all postponing related tests.
I assume that the firmware can keep three arrays of postponed hardware keyIDs, of their key press time points and of their release time points (which are NaN before a release has actually happened). The same info is also available for the key that triggered the macro; I refer to it by queue slot 0 below.
Based on these information, I think that the following grammar could unify `ifShortcut`, `ifGesture`, `resolveSecondary`, `resolveNextKeyEq`, the idea of this thread and probably many more scenarios:
```
CONDITION = {ifKeySequence | ifNotKeySequence} [[TIME_WINDOW]* INPUTEVENT]+

TIME_WINDOW = within MILLISECONDS [since {pressOf QUEUESLOT | releaseOf QUEUESLOT}]?
TIME_WINDOW = before releaseOf {QUEUESLOT | allPending} [plus MILLISECONDS]?
//If no time window is specified, use "before releaseOf queueKey 0" as default, i.e. wait
//for the specified INPUTEVENT as long as the macro trigger key has not yet been released.
//If no since clause is specified, count the time window from the last press in the queue.
//In case of multiple specified time windows per input event, they have implicit AND logic. 
//(OR logic can be implemented by multiple ifKeySequence lines in the macro.)

INPUTEVENT = [press | release] {not? KEYID | AnyKey} doNotConsume?
INPUTEVENT =          release   not? QUEUESLOT       doNotConsume?
INPUTEVENT = delay MILLISECONDS 

QUEUESLOT = queueKey <queuePos (NUMBER)>
MILLISECONDS = <time (NUMBER)>ms
```
Using this grammar:
- The current `ifShortcut` would simply be equivalent to `ifKeySequence  press KeyID1  press KeyID2...`, implicitly using the default time window `before releaseOf queueKey 0` when testing for each input event in the sequence.
- The current `ifGesture` would be equivalent to `ifKeySequence  within 1000ms since pressOf queueKey 0  press KeyID1  within 1000ms since pressOf queueKey 0  press KeyID2...`. 
- `resolveNextKeyEq` could also be superseeded by this syntax. For example, `resolveNextKeyEq 1 089 untilRelease...` would become `ifKeySequence  press AnyKey  press 089...`.
- If I understand the `resolveSecondary` description correctly, it should be equivalent to this OR combination of two ifKeySequence lines:
```
//Trigger by delay: If no release event and nothing else occurs for 350ms, then always trigger the secondary action. (But advance to the next macro line immediately on any other input.)
ifKeySequence  delay 350ms  exec secondary 
//Trigger by release order: If any secondary key is released no later than 175ms after the release of the primary key, then exec secondary. Do not let ifKeySequence consume any postponed events of the secondary key, i.e. of the hardware key ID that matches "not queueKey 0", because we want to play it only after a toggleLayer.
ifKeySequence  before releaseOf queueKey 0 plus 175ms  release not queueKey 0  doNotConsume exec secondary  
exec default
````
- Finally, the idea of a "virtual-mod-composed-of-two-physical-keys" from this thread could be implemented via: `ifKeySequence  within 200ms  press SecondTriggerKeyID  within 800ms before releaseOf allPending  press ThirdKeyID`.
- This grammar would also provide an easy way to combine this virtual-mod-composed-of-two idea with `resolveSecondary`: `ifKeySequence  within 200ms  press SecondTriggerKeyID  within 800ms before releaseOf allPending  press AnyKey doNotConsume exec secondary`.

I know that such a unifying/general grammar would mean a lot of work, but if you should ever consider refactoring/consolidating all the different postponing code scenarios... What do you think?

kareltucek, 2019-07-19:
Chapter One
=======

> The problem with delayed release events seems to be solved, too (hitting just cv or vc now only yields cv or vc after 2s, i.e. no OS repetitions like cccccccccccv etc. any longer).

Nope, I am pretty sure that this problem persists in one variant or another.

> holding a single key to trigger OS key repetition on purpose also works: cccccccccccccccccccccccc...

But only in the `Gesture` version, and with a delay of a second and half. Not very comfortable for daily use.

> Could ifShortcut get an option to keep listening until the release of the second key in the sequence?

Or more generally, listening as long as any key of the sequence remains pressed. Yes, can be done - see attachment and use `ifShortcutSeq` (and see `transitive` in the grammar below)

[fw.zip](https://github.com/kareltucek/firmware/files/3412765/fw.zip)

> Speaking of time windows, the user should get an easy "I changed my mind/cancel!"

The ability to "change mind" sort of makes sense, however, I have not encountered any need for it in my keymaps, so I tend to find it to be an "exotic edge case"...

I see that I should probably refactor the newly added abbreviations to by adding optional flags to `ifShortcut` and `ifGesture` instead of introducing new abbreviations. Something like:

    ifShortcut [OPTION]* [KEYID]*
    ifGesture [OPTION]* [KEYID]*
    OPTION = cancelIn <ms>   //this would consume presses if failed by timeout
    OPTION = timeoutIn <ms>  //this would end resolution and leave all presses in queue
    OPTION = transitive            
    OPTION = noConsume

Chapter Two
=========
Technically taken, it might even mean having to rewrite the postponer from scratch :-). Generally, I think that it is a quite nice draft which could work well if we really needed general queries into the queue. 

However, I am not convinced that we really need to. Lately, I've been trying to reduce amount of ugly commands in favour of easily usable abbreviations (like `ifPrimary`, `ifGesture`, etc), and I believe that more-or-less it is the good approach. I feel the proposed grammar to go against this effort. In the end, the simple abbreviations would have to remain for regular use, and the general grammar would serve for prototyping only. 

The grammar looks reasonable, but being reasonable does not make it less complex. E.g., I am not able to verify whether the proposed version of `ifSecondary` is correct or not.

orthoceros, 2019-07-22:
Well, complexity is a good thing, as long as its usage doesn't get too complicated. ;)  But I understand your language design goals and agree that the `ifSecondary` abbreviation is much more practical than writing "delay>350ms OR releaseSecondary<releasePrimary+175ms" every time in the above language...

Thanks for the attached firmware. `ifShortcutSeq` works, i.e. I can now use c->v->c and it keeps listening until v is also released. I agree, options would be more streamlined and more flexible than introducing another keyword for this postponing scenario. With your proposed options, I think the solution to this thread's virtual-mod-key-by-two-physical-keys idea could finally become:
```
$postponeKeys //### Macro for key c: ###//
$postponeKeys //Trigger: Test if c+v (here in the variant pressC->pressV) is pressed near-simultaneously; otherwise go to default:
                $ifNotShortcut timeoutIn 200 090 goTo default
$postponeKeys //Action sequences: Define the action keys to listen for and implement their actions:
                $ifShortcut timeoutIn 800 transitive 090 089 final tapKey LC-c  //(c->v)->c => copy
                $ifShortcut timeoutIn 800 transitive 090 090 final tapKey LC-v  //(c->v)->v => paste
$postponeKeys //Cancel: If the c+v trigger was held without any action key following before timeout, act as if the physical Mod key was held and released for some time (i.e. send nothing):
                $postponeKeys ifPlaytime 1000 final consumePending 1  //consume the second trigger key (v)
$postponeKeys //Default: "inject" a pressC at the front of the queue and then let all postponed keys be played back, like delayed normal typing:
                $default: postponeKeys final holdKey c
```
Some questions:
- Is the second `timeoutIn 800` counted with respect to the start of the second `ifShortcut` or with respect to the start of the macro? If it is not the latter, maybe another option like `maxPlaytime 800` could be added to avoid timeout stacking when testing for multiple sequences?
- Do I assume correctly that postponeKeys is not interupted between `goTo default` and `$default: postponeKeys`?
- Just out of curiosity: Is the playback of postponed release events guaranteed to always be in the physical release order? Because in code, the `holdKey c` possibly comes later. For example, I see the following `$default` cases:
  - within the 200ms timeout, another key is pressed that is not 090. No problem here (pressC->pressOtherKey followed by non-postponed releases in physical release order).
  - within the 200ms timeout, the macro trigger key is released: Then we should get pressC immediately followed by releaseC; no ambiguity either.
  - within the 800ms timeout and v pending in queue slot 1, c and v are released before a third press: Are both possible physical release orders of c and v preserved until after pressC->pressV?
  - within the 800ms timeout and v pending in queue slot 1, c is released first, followed by a third key (that is neither 089 nor 090): Here, I would expect that the OS sees pressC->pressV->releaseC->pressThirdKey followed by the non-postponed physical release order of v and the third key. Or would `holdKey` insert releaseC _after_ pressThirdKey in this case (even if there was a long physical delay between releaseC and pressThirdKey)?

kareltucek, 2019-07-22:
Wooa, you are always one step ahead of the intended usage :-). Lets number the lines:

    1:  $postponeKeys //### Macro for key c: ###//
    2:  $postponeKeys //Trigger: Test if c+v (here in the variant pressC->pressV) is pressed near-simultaneously; otherwise go to default:
    3:                  $ifNotShortcut timeoutIn 200 090 goTo default
    4:  $postponeKeys //Action sequences: Define the action keys to listen for and implement their actions:
    5:                  $ifShortcut timeoutIn 800 transitive 090 089 final tapKey LC-c  //(c->v)->c => copy
    6:                  $ifShortcut timeoutIn 800 transitive 090 090 final tapKey LC-v  //(c->v)->v => paste
    7:  $postponeKeys //Cancel: If the c+v trigger was held without any action key following before timeout, act as if the physical Mod key was held and released     for 8 some time (i.e. send nothing):
    9:                  $postponeKeys ifPlaytime 1000 final consumePending 1  //consume the second trigger key (v)
    10: $postponeKeys //Default: "inject" a pressC at the front of the queue and then let all postponed keys be played back, like delayed normal typing:
    11:                 $default: postponeKeys final holdKey c

- If cv is pressed, line three will consume* the "v" (the second trigger). I.e., you might want to add `noConsume` on line three, or to leave out `090`on lines 5 & 6 and change line 9 correspondingly .
- To implement the "cancel on timeout of pressed `cv`" could simply change the `timeoutIn` on 5 & 6 to `cancelIn` and remove line 9 entirelly.  
- Note that line 11 will postpone other keys until release of c. It should not matter while writting, but in case `c+<something>` was a shortcut in your software, it would prevent you from entering such shortcut. The `postponeKeys` should (almost) definitely not be present here.

*we expect that if `ifNotShortcut` is used, the code handling the shortcut follows and therefore consuming should still take place, just with opposite logic. Such handling makes sense if the final action does not fit one line.

I.e., the refactored result is:

    3:                  $ifNotShortcut timeoutIn 200 noConsume 090 goTo default
    5:                  $ifShortcut cancelIn 800 transitive 090 089 final tapKey LC-c  //(c->v)->c => copy
    6:                  $ifShortcut cancelIn 800 transitive 090 090 final tapKey LC-v  //(c->v)->v => paste
    11:                 $holdKey c

Questions:
- Timeouts are counted w.r.t. start of the macro (i.e., w.r.t. press of the activation key) unless `transitive` is present. If `transitive` is present, then w.r.t. last press event registered by postponer.
- Yes. Actually, the current implementation should even allow something like 2 cycles before postponer actually kicks and starts replaying anything, so you could probably leave out all the `postponeKeys`s in your macro.
- Well, yes, except the case when a macro itself actively triggers postoneKeys - in that case the macro actions work with *real* state instead of *postponed* state... which means that it may finish earlier than expected. Scenarios in order:
  - fine :-)
  - fine :-)
  - yes, as long as the `postponeKeys` on line 11 is removed, otherwise the result will probably be pressC->releaseC->pressV->releaseV
  - dtto - assuming that you remove `postponeKeys` on line 11, then yes, it should yield the "expected" result. Otherwise it may produce pressC->releaseC->pressV->pressThirdKey.

Also, I am not sure if I made that explicit, but the proposed grammar for `ifShortcut/ifGesture` options is already implemented and you are welcome to test it :-).

orthoceros, 2019-07-23:
...and you are always very fast and informative; thanks! The refactored/corrected macro version looks very good.

I thought that it wasn't implemented, yet, because I got ERR in the display. I still get ERR if one of the following lines comes first in the macro:
```
$ifShortcut cancelIn 800 transitive 090 089 final tapKey LC-c  //c->v ->c => copy
$ifShortcut transitive 090 089 final tapKey LC-c  //c->v ->c => copy
$ifShortcut cancelIn 800 090 089 final tapKey LC-c  //c->v ->c => copy
```
`$printStatus` says something like: `key c with shortcuts: 1: unrecognized command: transitive 090 089 final tapKey LC-c  //c->v ->c => copy` Only the version without any options works...

PS:
- Good to know that the postponer only plays back after 2 cycles. I sometimes just uncomment command lines in Agent during macro dev/testing and now can do this safely (as long as it is just a single comment line).
- Just to give you a cancel use case: I cancel usual pasting via LCtrl+V quite often when I search for some code expression, replace it with current clipboard contents, search next, replace again..., but do not want to replace all occurrences. Often I already have pressed LCtrl when scanning through the code, but do not want to proceed with +V to overwrite this occurrence; instead, I press the "search next" hotkey. So I need some cancel equivalent for the (c+v) mod key that does not overwrite the current selection with anything. However, I am also considering `(c+v)->space=>nothing` as a faster cancel-on-purpose option that does not have to wait for the timeout. 
- And/or even better: `(c+v)->n=>tapKey LA+downArrow //search next`! Having paste and search next actions based on the same virtual mod key should streamline code editing sequences like paste,paste,searchNext,paste,... even more. That would be cool! :) But I would need `noConsume` on the first two presses, `consume` on the third press (if it is an action key) and some `goTo backToStart` logic until triggers c and v are released both... And having c and v also enabled as action keys, if they come as third key press, makes it no less tricky, because their next press should then be interpreted as "regenerate the mod key" first. However, it should not be logically ambiguous, so it is doable in principal... (Anyway, this is for the future; need to work now ;)

kareltucek, 2019-07-23:
> I thought that it wasn't implemented, yet, because I got ERR in the display. I still get ERR if one of the following lines comes first in the macro:

You sure you are on current 8.5.4.kt.23 from standard releases? Implementing it was fast, but not that fast :-). 

> Having paste and search next actions based on the same virtual mod key should streamline code editing sequences like paste,paste,searchNext,paste,...

I guess you would have to give up those timeouts tough - because when doing replacing, you sometimes need to stop for some time and think, which would lead to accidents. Anyway in that case, I think that a new  command consuming nth pending would suffice. Then you could do something like:

    ifNotShortcut transitive 090 089 goTo next
    postponeKeys tapKey <whatever>
    postponeKeys consumePendingIdx 1 //the "third" key
    postponeKeys goTo 0
    next: ifNotShortcut transitive 090 90 goTo next //reusing labels is legal - in that case, jump goes one forward if possible, otherwise backward
    postponeKeys tapKey <whatever>
    postponeKeys consumePendingIdx 1 //the "third" key
    postponeKeys goTo 0
    //other cases...
    next: holdKey 089 //default action

Still, that's ridiculously overcomplicated... so we are getting back to the idea that it would be desirable if this "virtual mod" actually worked as a layer switch... I guess I could refactor those commands to something like

    holdLayer [HOLDFLAGS]* LAYERID
    holdKeymapLayer [HOLDFLAGS]* KEYMAPID LAYERID
    HOLDFLAGS = with KEYID
    HOLDFLAGS = max <time>

But in that case it is unclear how the activation should take place (e.g., who should consume the keys, and under which circumstances?; or how to prevent unwanted consumption of a key which is bound on the target layer?). And in that case you would still have to use one or even multiple `ifShortcut`s to ensure that hold is activated only under the strictly controlled conditions... Honestly the two-key-mod this still sounds to me as an overcomplicated and not-very-ergonomic idea.

Btw, is there any reason why you don't use the vim for your work? Because your motivations start sounding familiar :-). Vim or vim bindings allow exactly this type of streamlining, except that the editor "itself" switches between multiple modes depending on context of the workflow, so that you even don't need to hold the layer switcher.

orthoceros, 2019-07-26:
Oops, sorry; found and upgraded to kt.23. I am now using the following solution (same as above with a slightly modified cancel logic):
```
$ifNotShortcut timeoutIn 200 noConsume 090 goTo default  //look-ahead for the trigger. 

$ifShortcut transitive 090 089 final suppressMods tapKey LC-c  //c->v ->c => copy
$ifShortcut transitive 090 090 final suppressMods tapKey LC-v  //c->v ->v => paste

$ifPlaytime 1000 final consumePending 1  //always forget about normal typing, if we were in the virtual mod scenario for longer than 1 second, even if no action followed (the user changed his mind after holding the trigger).
$default: holdKey c
```
I actually like it! :) It seems to do everything reliably for single actions. (I could not provoke any typing interference or unintuitive delays, even ccccccccccccc works transparently.)
Just one thing: If `shift` is pressed while the trigger `(c+v)` is pressed and then released (long) before the action key, `shift` is still added to the `tapKey` action. Therefore, I added `suppressMods` to all actions. This is not ideal, as there might be actions that want to support shifting or other modding. Is there a way to only send those mod keys together with `tapKey` that are still physically pressed when the action key is pressed? If this conflicts with other scenarios and the mods are held on purpose in memory after their release, maybe a macro action `forgetAlreadyReleasedMods` could do the trick?
___
About vim: It is not really an option for me, as I need the IDE's integrated debugger, variable editor, code folding and so on. Besides, while certainly not as flexible as Vim, lots of Windows editors (my IDE, Notepad++, even MS Word) can be modified, for example, to "search next on Alt+downArrow". So, with my UHK as intelligent "gesture preprocessor", I could get streamlined editing gestures working in almost all editors I use!

However, I am open for more ergonomic gesture suggestions, of course! Besides the ability to streamline editing sequences, the trigger should be located near to the base row, not interfere with normal typing, the combos should be easily remembered and it should work quickly and reliably. Hmm. An alternative could maybe be "tap x+c+v within 200ms" as three-finger hold-nothing trigger => toggle layer EDT => map c to copy, v to paste, n to search next, ... => finally press Esc to toggle back manually? However, I fear that there are many editing scenarios that would require frequent switches of the context; for example, Ctrl+Shift+left/rightArrow to shrink/extend the selection before pasting. The xcv->actions->Esc sequences might be slower than standard hotkeys, unless all potentially related editing actions are available on this EDT layer. I still think that a "toggle back by trigger release" might be more convenient and faster. (Not sure, though. Will have to test any solution over a longer period in production anyway, once prototyping is finished.)
___
To enable consecutive action sequences, I agree that it would be more generic and easier to use a layer similar to the usual Mod key. What do you think about this design?:
```
$ifShortcut timeoutIn 200 noConsume 090 goTo triggerEditingContext
$default: final holdKey c

$triggerEditingContext: toggleKeymapLayer EDT base
$ifNotShortcut transitive noConsume 090 anyKeyWithDefinedAction goTo endHoldLayer
$postponeKeys executePendingAt 1  //<-This should consume the third key and call its action, as defined on the currently active layer (as if no other keys were postponed).
$postponeKeys goTo @-2
$endHoldLayer: untoggleLayer //(Does this also untoggle the keymap?)

//If the virtual mod scenario was active, then consume the second trigger key that is pending at index 0 and break:
$stillToImplementByARegisterFlag(ifExecutedAnyActionAbove) final consumePending 1 //if anything was executed above, we were not just "normally typing".
$ifPlaytime 1000 final consumePending 1  //always forget about normal typing, if we were in the virtual mod scenario for longer than 1 second, even if no action followed (the user changed his mind after holding the trigger).

$goTo default  //otherwise execute normal typing, after all.
```
Maybe `executePendingAt` can be realized by combining the existing firmware processor for non-postponed actions with the existing `call` logic for sub macros?

kareltucek, 2019-07-31:
> Therefore, I added suppressMods to all actions.

It suffices to add it to the `tapKey`s (etc). No need to have them at conditions... 

> This is not ideal, as there might be actions that want to support shifting or other modding.

Well, postponer was never meant to act to be used as a way to implement "hold" mechanisms, and I do not intend to be changing it... 

-----------------------
You know, most IDEs do support vim bindings to some extent... 

Regarding other suggestions:

I am currently testing vim bindings implemented on UHK, but its... imperfect. First problem is that every environment reacts slightly differently to the same key sequences. Second problem is that the keyboard has to guess some part of the information - e.g., whether something is selected or not.

But basically, same vim-like approach to streamlining could be used, just with differently defined operations. E.g., using left alt + command to switch into "command mode" and an edit command to go back.

Different thought regarding ergonomy is that I  have my keymaps set up to support "mirroring" via the caps-lock-positioned mouse button. This way, I can have right hand on mouse all the time and still controll entire keyboard solely by my left hand.

----------------------

I believe that this idea has the same problem with postponed shifts etc.. 

Furthermore, I don't like the part:

    $ifNotShortcut transitive noConsume 090 anyKeyWithDefinedAction goTo endHoldLayer
    $postponeKeys executePendingAt 1  //<-This should consume the third key and call its action, as     defined on the currently active layer (as if no other keys were postponed).
    $postponeKeys goTo @-2

If we want to hold a layer using active wait, I would much rather introduced `ifKeyActive KEYID`, so this part could be written simply as 

    $ifKeyActive #key goTo @0
    $ifKeyActive 090 goTo @0

> Maybe executePendingAt can be realized by combining the existing firmware processor for non-postponed actions with the existing call logic for sub macros?

I doubt it.

What about something like:

    $ifNotShortcut timeoutIn 200 noConsume 090 goTo default  //look-ahead for the trigger. 
    
    //check for all conditions which could lead to activation of the editing context
    $ifShortcut transitive 090 089 goTo secondary
    $ifShortcut transitive 090 090 goTo secondary
    $.... (any other ifShortcuts which should triggegr the thing)
    $ifPlaytime 1000 final consumePending 1
    $goTo primary
    
    $secondary: 
    $consumePending 1
    $final holdLayer with 090 <layer> 
    //holdLayer would be active for as long as the activation key or the `with KEYID` key is active
    //since postponed queue would be replayed here, the shift-related problem would go away
    
    $primary: holdKey c

Here `holdLayer` would simply hold for as long as the active or the referenced key was active, not consuming any presses.

 I think that this would be a syntactically sane way to do such thing. So this is option one and it requires only refactor of `holdLayer` to support optional arguments. 

However, it still does not solve some of your requirements in a reasonable manner. So I have a counterproposal no 2:

- add `ifKeyActive KEYID` which would check if a key is active at the moment (ignoring any pending events)
- add `ifKeyDefined KEYID` which would check if an action is defined, however on current keymap && layer only.
- add `%<idx in queue (NUMBER)>` which would expand into KEYID of nth press event in the postponing queue.
- add `activateKeyPostponed KEYID` which would add an activate and a deactivate event at the end of postponing queue.
- maybe add `ifKeyPending KEYID` later, just to keep consistency with `ifKeyActive`, which would look for the KEYID press event in the postponer queue.
- maybe rename `ifPendingId` to `ifKeyPendingAt`

Writing the functionality using these commands would still be ugly, however, it would give you a pretty powerful set of tools.

kareltucek, 2019-07-31:
Anyway if I decided to implement this whole thing as a single command, what would its syntax be?

orthoceros, 2019-08-03:
If you think that you can keep the maintenance overhead under control, having a complete (but minimal) set of generic commands for prototyping is a good idea, in my opinion. Especially, if you can internally reuse their code to implement higher-level commands. However, to avoid overloading novice users, maybe a sub-grammar of easy-to-use "everyday commands" should be presented first in the docu. (All prototyping/debugging/advanced grammar rules could be listed after that.)

>Anyway if I decided to implement this whole thing as a single command, what would its syntax be?

Ideally, commands are both flexible and easy in the default use cases. However, compressing the virtual mod idea into a one-liner that transparently consumes trigger keys with the fallback logic to normal typing was not as easy as I initially thought. Basically, I want the mapping "(macroKey->additionalTriggerKeys)->actionKeys => EDT.base.actionKeys". Finally, I came up with the syntax `route within 200ms 090 actionKeys to EDT base` using the following internal logic:

1) Look ahead for the trigger, here for press v within 200ms and then for press firstActionKey before all trigger keys have been released (this is the default time window; see below).
2) If the route is triggered, then toggle EDT.base. If EDT.base.firstActionKey resolves to the none action, toggle back and go to 3. Otherwise consume all trigger keys, call EDT.base.firstActionKey and continue holding this layer while any trigger key is still pressed. (Never return to the original macro in this case.)
3) If this route is not triggered or cancelled, do not consume anything and simply go to the next macro line. (Keep keys postponed for testing alternate route commands.)
---
The grammar could be as follows. If something is not specified in the command, defaults are defined by upper/leftmost OR-conditions in the grammar.
```
COMMAND    = route [TIMEWINDOW? keyId]* TIMEWINDOW? actionKeys 
             to [KEYMAP|current] [LAYER|current] 
             [cancel ifFirstActionKeyIsUndefined | never]?
             [until [allTriggersReleased minus <100ms safety margin> | manualToggleBack]]?
TIMEWINDOW = before [allTriggersReleased | macroKeyReleased]
TIMEWINDOW = within <200ms> [since [startOfMacro | lastKeyPress]]?
```
---
Logically, `resolveSecondary` followed by `holdLayer` seems to be a virtual mod with a trigger sequence of length 1 (just the macro key). I included the `minus <100ms>` in the grammar to reproduce the safety margin logic; is this the correct position in the code/time flow? Then `route actionKeys to current mouse` should work similar, except for the added cancel logic and the missing activation after a 350ms delay. (Why is an activation by delay needed here; can't we just always wait for the first action key before changing layers?)

kareltucek, 2019-08-04:
> Anyway if I decided to implement this whole thing as a single command, what would its syntax be?

Ewww. When I was asking this question, I meant to ask for a simple oneliner abbreviation, not for another proposal for complex all-encompassing grammar. 

> Logically, resolveSecondary followed by holdLayer seems to be a virtual mod with a trigger sequence of length 1 (just the macro key).

No, it is not. There is the fundamental difference in the resolution which you yourself originally pointed out in the old thread (which I believe is the only reason why we are having this discussion) - the smart secondary resolution resolves depending on *release* events, while shortcuts are resolved by *press* events. 

I'm probably getting lost, but I cannot see how these two different mechanisms are supposed to be both contained within the proposed solution without contradicting the need for virtual mod keys.

-----------------
Regarding the  350ms timeout, lets consider following scenarios:

Regular tap action on "secondary layer":
- layer switcher pressed (i.e., `resolveSecondary`)
- action key (e.g., `tapKey a` command) pressed
- user does nothing for a short time, *the firmware waits for release of one or both of the keys*
- one or both keys are released *(see next section for what can happen here)*
- *now the resolveSecondary resolution branches to either primary or secondary section and postponer starts replaying the events*

Hold action on "secondary layer":
- layer switcher pressed
- action key (e.g., `start mouse move left`) pressed
- user waits for the firmware to act without any intention of releasing the keys until something happens 
- *So, now the user waits for the firmware to move cursor left and the firmware waits for the user to release the keys. No action has been triggered so far.*
- *Obviously, we need to recognize this scenario somehow - thats where the 350ms timeout comes in*

-----------
Regarding safety margin, the actual release scenarios are these (the 350ms timeout never times out in these scenarios):

Switcher is released first:
- ...
- switcher key is released
- *firmware immediately recognizes this as two consecutive keystrokes and replays the primary actions*

Both keys are released at the same time:
- ...
- action key is released
- *firmware waits for safety margin (at most 50 ms)*
- layer switcher is released
- *Ok, lets see, the keys were released basically at the same time, this happens in writing scenarios - lets behave as if the layer switcher was released first, i.e., as in previous scenario*

Action key is released first:
- ...
- action key is released
- *firmware waits for safety margin (at most 50ms)*
- nothing happens in the meantime
- *firmware recognizes this as a secondary role scenario and therefore branches to the secondary section*

orthoceros, 2019-08-04:
> Obviously, we need to recognize this scenario somehow - thats where the 350ms timeout comes in

This is correct, if you want to use release logic also on the action key, not only on the trigger key(s). And yes, I suggested virtual mods to decrease interference with normal typing. But viewed on a timeline, even for this mousing scenario, the 1-key-virtual-mod seems to yield a similar attack surface with respect to interference than resolveSecondary. Lets compare them on a timeline:
```
//resolveSecondary; holdLayer; start mouse move left:
  pressMacroTriggerKey -> pressActionKey -> 350ms uninterrupted delay                  => start mouse move left
//route actionKeys to current mouse; start mouse move left:
  pressMacroTriggerKey -> pressActionKey -> 100ms safety margin before trigger release => start mouse move left
```
For the tapKey action, the timelines look like this:
```
//resolveSecondary; holdLayer; tapKey a:
  pressMacroTriggerKey -> pressActionKey -> releaseActionKey 50ms safety margin before releaseTriggerKey => tapKey a
//route actionKeys to current mouse; tapKey a:
  pressMacroTriggerKey -> pressActionKey -> 100ms safety margin before trigger release                   => tapKey a  (also supports holdKey a)
```
I may see this too simplistic or forget something. Maybe, longer delays than the 100ms default safety margin before trigger release are useful sometimes...but currently, I don't see it.

Of course, the interference probability of a 2-key-virtual-mod timeline `pressMacroTriggerKey -> pressAdditionalTriggerKey -> pressActionKey -> 100ms safety margin before all triggers are released => actions on target layer` is even "squared-lower" than both the 1-key-virtual-mod or resolveSecondary time lines...

kareltucek, 2019-08-04:
Ok, I think I now understand how you mean the activation logic. So lets try it in practice on 's' key:

    $postponeKeys ifNotPlaytime 100 ifNotPending 1 goTo @0
    $postponeKeys ifPending 1 delayUntil 100
    $ifNotReleased ifPending 1 final holdKeymapLayer MOU base
    $holdKey s

This is what you are proposing, isn't it? 

Doesn't work very well on my side.

-----------------------

Still, if we made parameters implicit, the minimum syntax for the oneliner virtual mod key could be 

     virtualMod [KEYID]+ KEYMAPID LAYERID

right?

orthoceros, 2019-08-04:
> This is what you are proposing, isn't it?

Almost, but the default time window between trigger key(s) and the first action key was `before allTriggersReleased`. So, for the 1-key-virtual-mod it should be (without the cancel logic on undefined action keys):
```
$postponeKeys ifNotReleased ifNotPending 1 goTo @0
$postponeKeys ifPending 1 delayUntil 100
$ifNotReleased ifPending 1 final holdKeymapLayer QWY mouse
$holdKey a
```
I actively use UHK mousing actions for quite some time now via the `a` key and `resolveSecondary`. I just switched to the macro above. The mouse starts moving snappier now! :) Other than that, I did not notice much difference, yet. Especially, I could not provoke any typing interference (asdasdsadasdasdasdasdasdasdasdsadasdsdasdasdasdasdasdasdasdasdasdasdasdasdasda... no missing a). Okay, if you hold `a` for long and then release it, nothing happened with `resolveSecondary` and now I get an `a` on release...but this could be easily solved via `ifPlaytime <cancelTimeout> break`. What exactly doesn't work for you?
Btw, it's cool that the language already supports enough prototyping flexibility for this trigger of the 1-key-virtual-mod! I did not think of `goTo @0`...it probably can make the CPU warm, but it works! ;)

Notably, holding a key for OS repetition does not work for the macro key in the 1-key-virtual-mod scenario (neither does it work with `resolveSecondary`). Of course, `holdKey` is a possible action for the actionKey on the target layer. In the 2-key-virtual-mod scenario, `holdKey` even works as action on macro key itself. Because there I use `within 200ms` as time window between the two trigger keys. This window is always met if I indent to use the virtual mod, because the gesture is to press both trigger keys simultaneously. The action key(s) then follow later, at any time before both trigger keys are released, minus the 100ms safety margin. If the 200ms trigger window is not met, the macro key can execute the default `holdKey` action, here.

---
>Still, if we made parameters implicit, the minimum syntax for the oneliner virtual mod key could be `virtualMod [KEYID]+ KEYMAPID LAYERID`, right?

Almost. If you want to include the 1-key-virtual-mod scenario (i.e. just the macro key as trigger), then the syntax should use `[KEYID]*` instead of `[KEYID]+`. With this minimal syntax you would also have to define different default time windows: First, `within 200ms since startOfMacro` between all trigger keys `[KEYID]*`. Then, from trigger keys to the first of the `actionKeys`, the time window should already be `before allTriggersReleased`. Otherwise, the 2-key-virtual-mod will not work correctly.

kareltucek, 2019-08-05:
First testing looked ok too, but then I tried some words which used to be problematic in the past and took days to figure out. For instance "user" (with secondary on 's') - simply doesn't work when written fast.

> it probably can make the CPU warm, but it works! ;)

No problem there - the language is interpretted, not compiled, and execution is done one command per cycle (in this context every execution is "different" command), so it works just as any other action - at most this probably reloads the macro action on every cycle. 

> Notably, holding a key for OS repetition does not work for the macro key in the 1-key-virtual-mod scenario

In the version with limited trigger time, it works, although I can see that it's probably not what you are after :-). 

> then the syntax should use [KEYID]* instead of [KEYID]+

True.

> First, within 200ms since startOfMacro between all trigger keys [KEYID]*. Then, from trigger keys to the first of the actionKeys, the time window should already be before allTriggersReleased. Otherwise, the 2-key-virtual-mod will not work correctly.

Well, experience with `resolveSecondary` and `resolveNextKeyEq` tells that specifying timeouts as part of commands is pretty annoying in the long run, so its better to find out which timeouts work and then hardcode them. Which also makes such thing much more usable for other users.

kareltucek, 2019-08-05:
But its true that the 1-key-virtmod will probably work well if the margin is increased to higher values (probably 200 or 250)

orthoceros, 2019-08-05:
> But its true that the 1-key-virtmod will probably work well if the margin is increased to higher values (probably 200 or 250)

Ahh, its not from writing too fast, but from writing too slow, then: "als als als als als s" Yep, reproduced (`a` is my trigger key and `l` has a mouse action on the target layer). How long do you need at maximum for releasing the `s` in `user` during normal typing? (It's a competition: If you increase the 100ms window, the mouse will start less snappy. So, for now, I will keep my version at 100ms, hoping that I will always release keys fast enough during normal writing...)

> No problem there - the language is interpretted...

I see; good to know!

> Well, experience with resolveSecondary and resolveNextKeyEq tells that specifying timeouts as part of commands is pretty annoying in the long run...

I agree in case of required positional parameters. But I would not recommend hard coding time windows: users have different writing styles (just like the release window above). So, a default that works for most out-of-the-box, combined with the flexibility of an optional parameter in the grammar to change this default for scenarios or users who want/need it, would be optimal, I think.

kareltucek, 2019-08-05:
150 does not suffice, 200 looks ok in testing. Cannot guarantee that it will be ok in normal writing. Reminds me of the way how I came up with the 350 - that was exactly the same "competition", where I started with 200ms (measured from macro key activation) and gradually increased all the way up to 350 whenever something misbehaved. 

I think that the sum of timeouts for both strategies will be almost the same at same reliability level. Your strategy is just a bit more "focused" in hold scenarios, so it will probably allow saving 50-100 ms on average. I think that in tap scenarios, the "release resolution" of resolveSecondary can actually act a bit faster under some circumstances (fast release & long timeout for reliability). I am now thinking about adding this logic it into the existing `resolveSecondary` logic to make the "350ms" timeout snappier... 

Still, I still dont feel strongly about implementing the actual virtmod since your expectations/requirements on the functionality are very specific and entagled with each other - it sortof makes more sense to provide just the basic prototyping commands anyway... 

> But I would not recommend hard coding time windows: users have different writing styles (just like the release window above).

I think that it might be better to set them by a separate global command (like `setSecondaryTimeouts 350 50 100`) if they are really needed.

---------------------------------------
Anyway, what are your current motivation for all this? (I suspect that part of the things which has been said over the last months is no longer relevant...) 

I mean specifically whether-or-not-and-why you are still after the 2-key-virtmod and whether-or-not-and-why does/doesn't 1-key-virtmod/resolveSecondary fit the bill?

(Please be brief! Or I'm gonna get lost again :D.)

kareltucek, 2019-08-05:
I'm pushing new release with the set of commands mentioned at the end of the Jul 31 comment...

orthoceros, 2019-08-05:
> Anyway, what are your current motivation for all this?...(Please be brief! Or I'm gonna get lost again :D.)

Okay, I'll try, but it's a long thread to summarize! ;)
- general motivation: gain speed and comfort during coding
- concrete motivation: get quick, reliable, easily memorizable, configurable command hot keys or sequences, without having to move fingers far from their normal base row positions
- original idea: quick gestures "hitBothAtAngle(c->v)->releaseBothSimultaneously => paste" and "hitBothAtAngle(v->c)->releaseSimultaneously => copy". 
- problems: These gestures could not be recognized reliably via a resolver logic by key release order and safety margin because of the simultaneous release of both keys at the end. Also, it showed bad interference with writing when trying to forward-detect the gesture via key presses only.
- virtual mods: Got the 2-key-virtual-mod idea of hitting both c+v (simultaneously in arbitrary order), then v for paste or c for copy. This trigger drastically reduced writing interference. It also allowed rescuing of the `holdKey x` default behavior on alphabetic keys.
- streamlined editing commands: Later, we wanted to support `hold EDT base` for command sequences in editing context (more than 1 action key after the trigger).
- cancel logic: Towards even lower interference with normal typing, I proposed a cancel logic for undefined action keys (in the layer case, cancel on all keys mapped to the none action).

Importantly, at the time of the original idea, I still used the default UHK mod row `Ctrl->Win->Alt->Fn->Mod`. With this layout, I found usual LCrl hot keys uncomfortable and slow for my little finger. Now I use `Fn->Ctrl->Win->Alt->Mod`, which feels much more like my original MS Natural keyboard. So, I am quicker now with usual hot keys. However, base row and quickness motivations still remain, just "less urgent". The additional motivation of gaining speed by streamlining typical editing work flows on a dedicated EDT layer came only much later, but I also see potential here.

Honestly, I currently cannot say, whether 2-key-mods and the cancel logic are really necessary with respect to further reduction of typing interference. If 1-key-mods worked sufficiently reliable, I would probably use them instead. Of course, they will never be completely transparent for all apps. Think of the holdKey default behavior and wasd-games, for example. (Then again, one could simply switch to a vanilla keymap for gaming contexts...)

---

> `setSecondaryTimeouts 350 50 100`

I am fine with global params. I could call an autostart macro with my user preferences on the `Space` key (and use a register to execute it only once per UHK boot...).
But why three timeouts? Where is the difference between the old safety margin (50ms) and the new "perform action, if triggers are not released in the next 100ms"? Would `setSecondaryTimeouts Inf 0 100` be equivalent to the latest prototype above? 
If I understand your logic `OR > 350ms since call of resolveSecondary` correctly, performing the first action no longer needs to wait for `100ms before release of all triggers`, if any trigger key is already held for long enough. This would allow for a snappier execution of the first action after holding triggers for sufficiently long, so I think it is a good combination. If you plan to extend `resolveSecondary`, what do you think of always using `setSecondaryTimeouts` (best combine all set commands with implicit postponeKeys) and then just `ifSecondary [withKeys [KEYID]*]?...`?

> ...it sortof makes more sense to provide just the basic prototyping commands anyway...

I am more than willing to have a few more lines in my macros, rather than hard-coded defaults or losing flexibility. After all, I only have to write a macro just once and then can duplicate it in Agent when needed. Once the macro works, I will probably never edit it again, anyway...

---

>I'm pushing new release with the set of commands mentioned at the end of the Jul 31 comment...

Thanks. Apropos prototyping: I have never tried to implement the original idea via a "postfix-trigger" based on the `releaseBothSimultaneously` part of the gesture. Of course, this gesture could never support layering and command sequences, but it would be interesting to study its typing interference. Is there some way to test whether a _postponed_ key has already been released? I think like this for key `v`:
```
$postponeKeys ifNotReleased ifNotPending 1 goTo @0       //wait for own release or next key press
$postponeKeys ifNotPendingId 1 089 goTo default          //cancel on undefined action keys (only allow c here)
$postponeKeys ifNotReleased OR ifNotReleased 1 goTo @0   //wait for own release or release of the postponed key (whichever comes first)
$postponeKeys delayUntil 100                             //wait for 100ms; this is the time window allowed for 'simultanous release'
$postponeKeys ifReleased ifReleased 1 final tapKey LC-c  //perform action, if both were released simultaneously
$default: holdKey v                                      //otherwise, perform default.
```

kareltucek, 2019-08-05:
> But why three timeouts?

It was meant more like an example, but yeah, it could be understand as an OR of all the timeout conditions...

>  If you plan to extend resolveSecondary, what do you think of always using setSecondaryTimeouts (best combine all set commands with implicit postponeKeys) and then just ifSecondary [withKeys [KEYID]*]?...

I've been thinking about something like that, but I am not sure that I can actually fit all the restrictions in there, so I do not plan to do that at the moment.

> Is there some way to test whether a postponed key has already been released? 

I should have seen this coming. Not at the moment.

kareltucek, 2019-08-05:
`ifPendingKeyReleased` is now implemented in 8.5.4.kt.25

kareltucek, 2019-08-05:
But regarding `ifSecondary [withKeys [KEYID]*]?...` yeah, I think I might do it eventually.

orthoceros, 2019-11-14:
Hi, first of all, here is some feedback based on my experience with virtual mods in the last months (sorry for not posting earlier; quite busy times...). 
- I am currently using virtual mods on `a` (for my arrowing.left layer), `space` (special keys layer), `;` (arrowing.right) and `Return` (mousing.right). 
- In this way, I can quickly and conveniently (i.e. without moving my hand from the base row) access my symmetric left- or right-hand layers.
- Even though being single-key virtual mods, typing interference using the following robust implementation is very minimal for me (you may need to adjust the delay to your typing style). Therefore, I gave up on multi-key virtual mods with practically zero interference chance for now, as they are just slower to type.
- Btw, `Ctrl-C` is now `a+g` or `;+h` and `Ctrl+V` is now `a+t` or `;+y`, i.e. included in arrowing layers.

The simple resolver implementation I use is still, for example:
```
$postponeKeys ifNotReleased ifNotPending 1 goTo @0
$postponeKeys ifPending 1 delayUntil 100
$ifNotReleased ifPending 1 final holdLayer mod
$holdKey a
```

Unfortunately, I have to report a regression: Today, I flashed 8.6.0.kt1 and all my virtual mods immediately caused high typing interference. I already tried to debug it. By using `delayUntil 1000` instead of `delayUntil 100` above, I found out that the `ifNotReleased` logic no longer works: Even if the key has been released long before `delayUntil 1000` expires, it still goes to the mod layer, instead of playing back the key press as usual typing. I have now gone back to 8.5.4.kt25 and all my virtual mods immediately started working again robustly. So it must have something to do with 8.6.0.kt1...any idea?

kareltucek, 2019-11-14:
Please, lets move bug report into a new ticket -> #40