kareltucek/firmware #32: Using a keymapping that uses control leaves control activated

VinceG3, 2019-06-14:
Version 8.5.4 of the official firmware fixed an issue where:

- Make the modifiers of shortcut keys stick only as long as the layer switcher key of a shortcut key is being held.

However this version of the firmware still exhibits the issue. To reproduce, remap a key to control anything, it most annoys me in a browser, so try control-w for close tab. Close a tab with the remapped key, then click on any link. It will control-click the link.

Flashing back to latest vanilla fixes it, while flashing back to 8.5.3 brings back the issue.

kareltucek, 2019-06-14:
Thanks for report, will look into it soon! Unless you need sticky modifiers, you should be able to disable them entirely by $setStickyModsEnabled 0.

VinceG3, 2019-06-14:
Setting Sticky Mods off got rid of the behavior for me, thanks! Didn't realize that was a feature people actually needed, curious what it's used for.

kareltucek, 2019-06-14:
It's for windows's alt+tab, but the proper behaviour should definitely be that of the official 8.5.4 ;-).

kareltucek, 2019-06-28:
Fixed. Thanks for the catch!

closed 2019-06-28 by kareltucek