kareltucek/firmware #33: Custom "live" strings in macros

pixel217, 2019-06-17:
I would like to have a "record_keys" and a "playback_keys" command in the macros that would allow capture and use of "custom live strings" to enable such usecase :

1 start the macro
2 record_keys (user types "hello" followed by a termination command eg. Mod+Enter)
3 other macro commands
...
...
n playback_keys (would type "hello" where the cursor stands)
n+1 other macro commands
end of macro

The system described here is a single "custom live string" buffer but with labels "record_keys [StringName]" "playback_keys [StringName]" it could be extended to more than one "custom live string" buffer. I don't expect the "custom live strings" to be persistant between macro runs or different macros.

Sorry if the format of this "idea" is not the usual one, I try to contribute an idea that is a bit more ambitious than I am ready to implement myself in this codebase....

kareltucek, 2019-06-17:
I am not entirely sure whether I understand correctly what you mean, but I think that this can be already done via "runtime macros".  

Lets assume two macros:

    $recordMacro .
    $ifRecording goTo @0
    <do something ...>
    $playMacro . 
    <do something ...>

and:

    $stopMacro

The `recordMacro` will start recording into a register identified by '.' (a character, not a string). Then, the macro waits for as long as the recording is in progress. When the user stops recording by another macro (by `stopMacro` command), the macro resumes.

kareltucek, 2019-06-17:
(The `@0` is a relative address denoting "go by zero commands", i.e., to the same one, effectively acting as an active waiting loop.)

pixel217, 2019-06-17:
thanks for the feedback, as soon as I get my UHK (probably end of week or next week) I will try the example you gave here to validate if it corresponds to what I tried to express here. I will of course update the "issue/idea" in consequence.

pixel217, 2019-06-24:
Hello, typing from my brand new UHK (well not brand new... I flashed it with the current firmware already ;-)

and I must say that your explaination here-above fits my needs except for one little thing : while recording, I would like the scancodes not to be sent to the system (blind recording if you prefer) because for instance I could be typing a filename and having the followup of the macro save multiple files using said filename but I dont want the app to react to the keystrokes of the app name should those be linked to app functions.

Is this possible? (I am diving in the documentation of the firmware as of now to check but maybe you will be faster to answer ;-))

kareltucek, 2019-06-24:
Hi!

You are right, I've overlooked it. I will look into it and hopefully introduce some solution.

pixel217, 2019-06-25:
I guess some kind of "$echoOff" "$echoOn" would be sufficient

kareltucek, 2019-06-25:
It would, but it would be pain to use... I am more inclined to doing a slight refactor of recording commands and adding versions for blind recording just as you suggested, which would suppress sending scancode reports and moreover block until recording was stopped...

Either way, it will have to wait a few days (till weekend maybe?). I understand that it may be hard provided that you've just received your UHK, but please, be bear with me :-). Or if you know a bit of C, you can try messing with firmware yourself in the meantime :-).

pixel217, 2019-06-25:
I actually am a software engineer so I would but I have yet to find where in the code such commands are being interpreted.

don't worry for the wait, I am just trying to provide as much input as possible to help the development of this firmware for this great piece of hardware.

kareltucek, 2019-06-25:
> I actually am a software engineer...

Basically, in `macro.c`. In more details, there's an 'Adding new features' section in the readme of this firmware, which will tell you more, and of course you are welcome to ask for more details. Regarding this issue - probably leave it to me, since I know what I want to do with it ;-), but there's lots of other things which could be done :-).

kareltucek, 2019-06-28:
Here we go, just added `recordMacroBlind`, as well as `startMacro` and `startMacroBlind`, (still non-blocking though... not sure if blocking or nonblocking is more intuitive). 

I am closing the issue, however feel free to follow up here.

closed 2019-06-28 by kareltucek

pixel217, 2019-06-28:
Not sure what startmacroblind / playmacroblind are supposed to do/be used for?

kareltucek, 2019-06-28:
Sorry, its `startRecording` and `startRecordingBlind` :-). (I've added them so that there is an alternative which explicitly "starts" recording instead of "toggling" recording. )