kareltucek/firmware #38: Can't get the Double Tap to Caps Lock example to work

nach00, 2019-10-26:
I've been trying for hours to figure out how come I can't get the example Double Tap to Caps Lock example to work! I've tried both the new version and the old version. Oh, I'm also on a Mac.

I'm fairly certain I'm doing everything correctly because I created another macro that works fine (Mod+C to close tab, Mod+Shift-C to reopen tab).

I thought it might be a conflict with my Better Touch Tool, but I disabled the software and it's still not working. 

The Shift part works. But the Double tap does not. 
I've even tried on a different key other than Shift, and Double tap still doesn't work.

[image]
[image]

kareltucek, 2019-10-26:
Hi! Mac is here to blame. The system requires caps lock to be pressed for certain amount of time in order to work. The tapKey however activates only for a very brief moment.

So you will need:

$holdKey leftShift
$ifNotDoubletap break
$pressKey capsLock
$delayUntil 500
$releaseKey capsLock

Try it and if it works, then lower the 500 ms timeout to something more reasonable (100/200/300 maybe).

Also in case you are using Karabiner, it is known to cause problems in combination with some features of the UHK.

kareltucek, 2019-10-26:
I've added note about this into the readme, so everything should be clear now -> closing the issue, but feel free to follow up.

closed 2019-10-26 by kareltucek

nach00, 2019-10-26:
Works! Used 100

kareltucek, 2019-10-26:
:-)