kareltucek/firmware #41: Can't make suppressMods work on macOS

KeyAmam, 2019-12-08:
Hi, I can't make `suppressMods` work on macOS 10.14.5

1. If you're using Karabiner Elements on your Mac, close it!

    Done.

2. Install the latest Agent:
   https://github.com/UltimateHackingKeyboard/agent/releases/latest

    Done.

3. Use Agent to update to the latest firmware:
   https://github.com/UltimateHackingKeyboard/firmware/releases/latest

    Done.

4. Try to reproduce the issue, and only report it if it still persists.

  - Create a macro:
    ```
    $ifAlt suppressMods tapKey a
    ```
    ```
    $ifNotAlt tapKey b
    ```

  - Assign the macro to a key

  - Press `a`
    Ouput: `b`

  - Press `rightAlt + a`
    Output: `å`

Am I missing something?

kareltucek, 2019-12-08:
Hi!

You are totally right. Good catch. Will fix it soon.

kareltucek, 2019-12-08:
Ok, I see where the problem is, but it will probably take some time and thinking to fix. For the time being, it should suffice to add `$noOp` as first command. 

The problem happens because the first command is executed on key activation (i.e., in the middle of an event update cycle)(the reason is to prevent races on postponing if I remember correctly). Regular way is to execute macro commands before other event handling. So what is happening is that the modifier is written into output report before suppressing flags gets toggled. It will happen for macros whose keyid is higher than keyid of modifiers, which is probably why nobody noticed so far - left side modifiers have highest keyids of all keys on the keyboard.

KeyAmam, 2019-12-09:
Thanks, I understand why and `$noOp` workaround works perfectly.

KeyAmam, 2019-12-09:
One more thing I would like to do but `noOp` seems not to work in this case.
```
$ifAlt suppressMods tapKey semicolonAndColon
```
```
$ifNotAlt tapKey a
```
with rightAlt
Expected output: ;
Actual ouput: …

without rightAlt
Expected output: a
Actual output: a

Any workaround?

kareltucek, 2019-12-09:
I can't reproduce that (linux here), which means that the problem is probably not in the firmware.

- Is `alt+; -> ...` a global binding on mac?
- Can this behaviour be application specific? (Again - Karabiner or any other keyboard preprocessors could be the culprit.)
- You can try adding a delay like `ifAlt suppressMods delayUntil 50` so that OS sees some delay between release of alt and press of `;`.
- If you post your UserConfig, I will have a look at it.

KeyAmam, 2019-12-09:
Stupid me, Karabiner was running...
It works fine. Thanks!

kareltucek, 2019-12-09:
:-)

kareltucek, 2020-07-05:
Fixed!

closed 2020-07-05 by kareltucek