kareltucek/firmware #44: interrupting running macro

bistabil, 2020-01-20:
Is this possible?
Since macros are executed asynchronously is there any way to stop them?
My use case is following:
I want to repeat some keystrokes indefinitely or until I interrupt it manually.
So far I've managed to repeat keystrokes indefinitely by reseting register's state.
Actually I can interrupt it by powering off the keyboard but that's only way I've found.

kareltucek, 2020-01-21:
> So far I've managed to repeat keystrokes indefinitely by reseting register's state.

I don't understand this, would you clarify?

Regular way would be to use `$goTo`
E.g.

    $tapKey a
    $tapKey b
    $goTo 0

> Since macros are executed asynchronously is there any way to stop them?

Yes, you can signal using a register. 

E.g. InterruptionKey:

    $setReg 5 1 

Macro:

    $setReg 5 0
    $tapKey a
    $tapKey b
    $ifRegEq 5 1 break
    $goTo 1

Also, there is a set of diagnostic commands which reset the keyboard state including hard interrupt of all running macros. But you do not want to use that.

bistabil, 2020-01-22:
I will test this today.
So if I understood it, registers are like "singletons" and their state is shared between different macros?
I didn't expected this to be the case as macros themselves seem not to behave as singletons and you can execute the same macro multiple times in parallel so I thought that registers could also be "created" for each macro instance.

kareltucek, 2020-01-22:
Correct. "registers" are simply global variables.

kareltucek, 2020-01-26:
Also, thinking about it again, the shortest variant is probably to use the conditional for modifiers. This variant would not need any signaling via registers...

E.g.:
$tapKey a
$tapKey b
$ifAnyMod break
$goTo 1

bistabil, 2020-03-14:
And how about longer action sequences?
Let's say that there are a lot of actions before $ifAnyMod break - commands such as delays and mouse movements. Is there another way to stop it other than using $ifAnyMod break after every action?

kareltucek, 2020-03-14:
There is `diagnose`, which will however do more than that and should not be used as a regular way. 

But I think I can add a new command like `stopAllMacros`. However, it might have to wait until situation around secondary roles is resolved.

kareltucek, 2020-03-14:
Done. 

Could you please test that it works as expected?

bistabil, 2020-03-14:
I will test it now.

kareltucek, 2020-03-14:
So I guess this can be closed. (Ofc., feel free to follow up.)

closed 2020-03-14 by kareltucek