kareltucek/firmware #45: writing value of register

bistabil, 2020-03-14:
Is it possible to write value of register?
Something like
$setReg 0 10
$write #0 -> should result in 10 being output

Simplest use case:
Output sequence of numbers like 1,2,3,4,5..
Why not just create macro containing write commands for each number in a sequence?
1. There are a lot of numbers in this sequence so it would be nice to reduce number of commands in the macro
2. My use case is not simplest one, it actually includes 60+ commands that would have to be repeated with only one of them being write command with variable number

What I made so far:
I've copied the macro 40 times (because of sequence of 40 numbers) and hardcoded values inside of that write command but 2400+ commands are making agent less responsive and if I have to change one thing in that 60 actions requence I'll have to reapply it 40 times

How I could improve it:
Enclose part that has to be repeated within a loop.
Then for that 1 write command with variable part I should create 40 different write commands and use condition (check register equality) to trigger one of them in each iteration of a loop
In this way, I could prevent copying of rest of the macro

kareltucek, 2020-03-14:
Well, I never thought there would be an actual use case for this, apart from debug reasons. Still yes, I needed this feature before for debug reasons, and it is available - just not documented and not entirelly straightforward. 

There are actually two ways to output something: 
- Directly by `write` command. This directly gives an address of a string to an algorithm which sends the scancodes. 
- To write things into "debug buffer" first, and then print the debug buffer. E.g.,  `$setReg 5 7; $setReg 0 5; $setStatus Register number #0 contains ##0; $printStatus` should write out `Register number 5 contains 7`. 

The only trouble is that `setStatus` automatically appends a newline at the end of your string... I guess you would prefer to be able to print without such newline?

bistabil, 2020-03-14:
I am not sure what you mean by "address of a string to an algorithm"
About the second solution, in my concrete use case new line wouldn't create an issue as I am outputing it to single line input that will ignore it.
I will test second approach now.

kareltucek, 2020-03-14:
> I am not sure what you mean by "address of a string to an algorithm"

Nevermind :-). 

> When printing status having value of register in it I see that also one space seem to be prepended to the value.

Forgotten about it. Yes, makes life easier. I will remove it from the interpolated command, just give me a minute.

kareltucek, 2020-03-14:
Still not perfect, but should suit you better now.

kareltucek, 2020-06-03:
I have fixed some problems related to the `setStatus` command (so that now printing (e.g.,) a number in apostrophes should work), and adding a new `writeExpr` command, which will print result of a numeric expression (e.g., `#5`).

(I am not implementing classical string interpolation in `write` command for technical reasons.)

closed 2020-06-03 by kareltucek