kareltucek/firmware #47: I18N question: RA-8 is displayed in Agent, but $printStatus reports

jscheidtmann, 2020-03-19:
I am trying to solve my issue with a german keyboard layout on Windows 10 using your modified firmware.

On a german keyboard layout in Windows, the {[]} characters can be entered by pressing

"RAlt"+"7","8","9" and "0" (on the number row).

I now have created an openKlammer macro that reads in Agent:

> $ifNotShift suppressMods tabKey RA-7
> $ifShift suppressMods tabKey RA-8
> $displayStatus

and similar for closeKlammer but with RA-9 and RA-0. These are now assigned to the keys "[" and "]" (as given by the American keycaps). When I now press the key I get error messages, which are displayed like this:

`openKlammerÖ 0Ö unrecogniyed commandÖ tabKey RAß8`
for pressing "["

and 

`OPENKLAMMERÖ !Ö UNRECOGNIYED COMMANDÖ TABKEY RA?/`
for pressing Shift+[. 

Note that on a German keyboard "y" and "z" keycaps are swapped and ";" is ö, "-" is ß. 

Entering `tabKez` instead does not work.

What would be your expectations? And what should I enter? 

Thanks in advance, Jens

kareltucek, 2020-03-19:
Should be `tapKey`, not `tabKey` :-)

jscheidtmann, 2020-03-19:
*blush*

jscheidtmann, 2020-03-19:
Thanks for pointing the obvious out.

closed 2020-03-19 by jscheidtmann

kareltucek, 2020-03-19:
No problem!