kareltucek/firmware #48: [MINOR] Please add iso key to KEYABBREV

jscheidtmann, 2020-03-20:
I have got the UHK in ISO layout, that means left-shift keycap is split and there is an additional key. The keycap displays "\ |", in agent, the scancode is "| ISO". 

In the grammar you list almost all keys in KEYABBREV, but the ISO key seems missing. Through trial and error I have discovered that nonUsHashmarkAndTilde seems to be the right thing to use.

I expected that KEYABBREV mirrors what's available in Agent, if only for consistency reasons. 
Please add "pipeIso" (or some such) as a synonym for nonUsHashmarkAndTilde, if that is the right key.

kareltucek, 2020-03-20:
> I have got the UHK in ISO layout, that means left-shift keycap is split and there is an additional key. The keycap displays "\ |", in agent, the scancode is "| ISO".
>
> In the grammar you list almost [...]

Sure, will look into it later today or tomorrow.

> I expected that KEYABBREV mirrors what's available in Agent, if only for consistency reasons.
> Please add "pipeIso" (or some such) as a synonym for nonUsHashmarkAndTilde, if that is the right key.

No. They mirror source code definitions given in `./right/src/lufa/HIDClassCommon.h`. I converted it directly from there and just cleaned up some which were obviously misleading. The mapping itself is hardcoded in './right/src/macro_shortcut_parser.c'. In case you want to dig into it deeper, feel free to do so and post a PR :-).

kareltucek, 2020-03-22:
There are actually three different scancodes - `backslashAndPipe`, `nonUsBackslashAndPipe`, and `nonUsHasmarkAndTilde`. 

According to `keymap.c`, the Agent `| ISO key` should correspond to `nonUsBackslashAndPipe`. I am not entirelly sure what `nonUsHashmarkAndTilde` is, but on my side it behaves as `backslashAndPipe` (both produce `\|`). On the other side, `|ISO`/`nonUsBackslashAndPipe` produces `<>`. 

I am adding `backslashAndPipeIso` alias, which should make things a bit clearer.

Closing, but feel free to follow up.

closed 2020-03-22 by kareltucek