kareltucek/firmware #50: Make sticky keys more configurable

kareltucek, 2020-05-28:
Followup of UltimateHackingKeyboard/firmware#299 .

kareltucek, 2020-05-28:
> Let me know if the initial post remains unclear.

@matthewwardrop I am afraid I am still unclear on the key-related notation you are using. I still see things like `press + <Key mapped to >` or ``` + <key mapped to + <`>>```, where at most one key of a shortcut  is clear. I guess that you probably mean specifically the `super + backtick` combination (mentioned in gnome docs), but from lack of any mention of the `super`(/`gui`/`win`) modifier it could as well be that you are trying to use the backtick itselfs as a modifier (which would be a problem).

matthewwardrop, 2020-05-29:
Let's try again :). I want the <Alt> + <grave/backtick> shortcut to be sticky, as is <Alt> + <Tab>, when simulating those key presses using <Mod> + <g>; where g here makes sense because it is right next to the default window switching shortcuts in the default mod layout. The solution seems to be either to hard-code some extra keycodes as sticky, or to make it configurable per key mapping. I understand the latter is probably a large chunk of work, and so I'd be happy with hard-code keymaps, and even, if necessary, to compile my own version of the firmware with these changes (that I would make myself). If this is not a highly sought after feature, then I would not want you to waste your time generalising things unnecessarily.

(And thanks for your explanation re:the nature of this repo. I think I have a similar scratch-the-itch approach in many of my projects, but doubt I will have a lot of time to devote here. If that changes, perhaps you can expect some PRs). ;).

kareltucek, 2020-05-29:
Alright, I see now.

From my point of view, the possible solutions are:
- To hardcode it. If you prefer this variant (as a way of becoming acquainted with the codebase), then it suffices to add the shortcuts into https://github.com/UltimateHackingKeyboard/firmware/blob/2f7837d83927695ec1e742abb9050e8adac1e1b5/right/src/usb_report_updater.c#L103 .
- Otherwise (and most likely anyway), I think I will extend the tapKey, holdKey and pressKey commands by an optional `stick`/`sticky` flag, so as a result stickiness could be achieved on per-key basis via macros like `$holdKey stick LA-graveAccentAndTilde`.

matthewwardrop, 2020-05-29:
That sounds great! Thanks @kareltucek .

For my edification, what are the scancodes that are actually sent on the USB channel for a sticky key with a <alt> + <tab> like keystroke? I presume it is something like:
- <alt> down
- <tab> down
- <tab> up
- time passes
- <alt> up

And so with the <mod> + <g> keystroke, we are tying the <alt> events with the mod key strokes; such as:
<mod> + <g>
- <alt> down
- <grave> down
release <g>
- <grave> up
hold <g>
- <grave> down
release <g>
- <grave> up
release <mod>
- <alt> up

Is that right?

matthewwardrop, 2020-05-29:
Interesting. Is that behaviour unique to your firmware? I don't see that behaviour on stock firmware. When I press <Mod> + <D> (<Alt> + <Tab>), and then release <D>, the app switcher does not disappear even when I press space; only when I release the <Mod> key. Perhaps the sticky behaviour overrides it?

kareltucek, 2020-05-29:
Interesting... is your space bound to a scancode action on the layer where you are testing it?

The behaviour should be the same on both firmwares.

matthewwardrop, 2020-05-30:
Oh... whoops! I had completely forgotten I'd remapped the Space bar to <Super> on the mod layer. That makes a bit more sense.

matthewwardrop, 2020-06-03:
Nice... Thanks @kareltucek !

matthewwardrop, 2020-06-13:
Finally had a chance to play with this, and it works beautifully (once I figured out how to use macros)!

Thanks again!

kareltucek, 2020-06-13:
Glad to hear!