kareltucek/firmware #51: Output higher ascii char on keyboard shortcut?

highend, 2020-09-30:
Is it possible to let the keyboard output (modifier keys = output char):
WIN + ; = ö
WIN + SHIFT + ; = Ö
WIN + , = ä
WIN + SHIFT + , = Ä
WIN + - = ß
etc.
with this firmware?

If yes, could you show me how, please?

US ANSI UHK keyboard, US keymap loaded (on Windows)

kareltucek, 2020-10-04:
Hi!

So, when you press physical keys `WIN + ;`, then you want the keyboard to produce some sequence of scancodes which will be interpretted as `Ö` by your operating system, right?  (Which sequence is it?)

Is this right?

If so, and if the number of keys is small, then you can use `ifShortcut` command. E.g.,

```
$ifShortcut 18 final tapKey RA-semicolonAndColon
$ifShortcut 85 18 final tapKey RALS-semicolonAndColon
...
holdKey LG-
```

- The macro is to be bound on the WIN key
- RA = right alt
- LG = left gui (= WIN) 
- I am assuming that the `ö`/`Ö` keys are produced by right alt + semicolon keys, since you do not specify...
- `final` terminates the macro after the command
- `94 64` are keyids identyfying physical keys on your keyboard (85 = left shift, 18 = semicolon in standard us layout). You can find out the keyids  using the `resolveNextKeyId` command (i.e., use the command in a macro and then press some key, and the keyboard will print out the number (using enUS scancodes, so you need to use enUS layout)).
- semicolonAndColon is a key abbreviation - you can find available abbrevs in the readme's grammar.

If the number of such mappings is higher, then it is a better idea to create a new "WIN" layer/keymap. Then you can use the `holdKeymapLayer` to access it (so that it does not eat up one of the four basic layers).

highend, 2020-10-16:
Hi,
sorry, was on holiday (still possible even in Covid times...).

It's not about scan codes. At least not, if I understood your explanations correctly. The US keymap that I'm using on my Windows OS is not aware of german umlauts so sending a scancode from my UHK would just result in that key being printed. Alt codes would be necessary.

```
Ä = 0196
Ö = 0214
Ü = 0220
ß = 0223
ä = 0228
ö = 0246
ü = 0252
```

kareltucek, 2020-10-16:
Hi! 

Well, there is no special support for alt-codes/unicode-chars in this firmware, but it still should be capable of producing the alt shortcuts. Something like:

```
$pressKey RA-
$suppressMods ifShift  writeText 0196
$suppressMods ifNotShift writeText 0228
```

This should produce `<right alt> + 0 + 1 + 9 + 6` with shift and `<right alt> + 0 + 2 + 2 + 8` without shift. You could then bind macros like this on a dedicated layer, and make the layer accessible via the WIN key.

(Remapping just those ~7 mappings on WIN would be complicated since the language does not support scoping.)

highend, 2020-10-16:
Hi,

are you sure that this should work?

I've created a macro, named it "ö"
Content is a single write text action with these 3 lines:
```
$pressKey RA-
$suppressMods ifShift  writeText 0214
$suppressMods ifNotShift writeText 0246
```
Saved the macro and bound it for testing purposes to the "fn" layer, to the key ";" (and ofc saved it to the UHK afterwards).
I don't get any output in a text editor when I now press fn+;

Normally these alt codes require the number keys from the numblock, the normal 0-9 keys can't be used for this kind of stuff.

I also tried to add a leading "Tab key: Caps Lock" and the same as the trailing (last) macro action but to no avail...

kareltucek, 2020-10-16:
> Normally these alt codes require the number keys from the numblock, the normal 0-9 keys can't be used for this kind of stuff.

Totally forgotten about numeric block :-(. 

> Content is a single write text action with these 3 lines:

You would need three text actions with single line commands, but as noted above, alt codes will not work with normal numbers...

Then I am afraid that it is not possible within a reasonable amount of effort atm.

highend, 2020-10-16:
I tried it with 3 single lines as well :-)

Too bad that it's not possible!
It's the only thing I'm really missing on my UHK.

Thanks for your help!

GrauLab, 2020-12-06:
Well, I have solved this problem a while ago. The solution has two parts. In this first post, I only describe the "narrow" solution of just entering a single Unicode character via right-alt mapping. I only still use it for entering ß in Outlook, because MS Outlook has some weird hardcoded hotkey on RA-s, which would be the normal way of entering ß using the general solution below (see my next post). 
So, here is the code for my "fix=entering ß in Outlook via Unicode escape" macro. It is bound to the s key on my custom ALT keymap:
```
$ifShift final holdKey RA-s //only do something on non-shift, as I want RALS-s to still yield § (see next post).
$tapKey lockingNumLock //see remark below

$pressKey leftAlt //start Unicode alt sequence
$tapKey keypad0AndInsert //tap the code for the char you want on NumPad
$tapKey keypad2AndDownArrow
$tapKey keypad2AndDownArrow
$tapKey keypad3AndPageDown
$releaseKey leftAlt //terminate Unicode alt sequence
```
One remark about `$tapKey lockingNumLock`, though: This will have side effects on apps like Notepad++ that listen and change their app state on this key. There does not seem to be an `ifIsLocked NumLock` and neither a `$tapKey unlockNumLock`. Maybe it is better to just map a key in Agent to `NumLock` and make sure manually that it is locked when you need it (or once after boot if you never use NumLock keys for anything else).

kareltucek, 2020-12-06:
> Well, I have solved this problem a while ago.

Yes, this works fine for a small number of special characters. However, I am quite doubtful about anyone wanting to map a full language set this way (e.g., czech has 15 such characters, with different lower/upper case codes, which (including shift-sensitive setup) means 45 macroes). There definitely should be a better way to achieve it (which is why I haven't closed the ticket so far - I would actually like to implement some reasonable solution, just haven't figured out the ideal design).

> There does not seem to be an ifIsLocked NumLock and neither a $tapKey unlockNumLock.

 That's true. I did not have a reason to do this yet. Atm, I have no idea whether the UHK actually knows the state (although it is reasonable that it does). Will probably look into it when I am working on the code again.

GrauLab, 2020-12-06:
Okay, now some details about the general solution I currently use. Windows maps scan codes that it receives from your UHK to characters via a keyboard layout. So the idea is to change the keyboard layout to match your needs, ideally reusing an already existing one.

I have long searched for a layout that suits my needs, in particular one that contains all German umlauts without defining dead keys, because the latter are really annoying while coding. In the end, I defined a new one myself, but based on the official English US international QWERTY keyboard layout (with dead keys) that you can also simply add via the Windows settings GUI. 
I did not remove any characters "stored" behind dead keys, but they are now only dead if pressed in combination with `RA`. For example, on `RA-'` Windows waits for the next key and if it is followed by `a`, then you get `á`. [Credits: I was inspired by [github.com/thomasfaingnaert/win-us-intl-altgr](https://github.com/thomasfaingnaert/win-us-intl-altgr), but for some reason I could not select his layout in the GUI after installing it, so I re-created it starting from a clean export of the current official US intl layout.]

Installing a custom keyboard layout in Windows is a bit complicated, but you only have to do it once:
1) First, store the attached definition file [KBDUSX_WS2019_MICmovedDeadKeysToAltGr.klc.txt](https://github.com/kareltucek/firmware/files/5648790/KBDUSX_WS2019_MICmovedDeadKeysToAltGr.klc.txt) as `.klc` file on your disk. [I.e. remove the `.txt` suffix that I had to add for this Github upload. Also note that I had to enter something in the COPYRIGHT field of the klc file, but from my side feel free to ignore it.] 
2) Then download and install the Microsoft Keyboard Layout Creator 1.4 and open the `klc` file with it. It is an old tool, but it still works well, even in Windows Datacenter Server 2019. In the program, use the "Project" menu and select "build DLL and setup package". (Maybe you have to install the .NET runtime first; don't recall.) Then double click the generated msi installer file for your CPU bitness, likely `_amd64`. 
3) After installing, go to Settings\Time & Language\Language, select "English (United States)" and click the `Options` button. (If you do not have "English (United States)" in your language list, you first need to add this language, as the `klc` file is only defined for `LOCALEID "00000409"`). In these language options, click "Add a keyboard" and then choose the new entry "US Multinational - dead keys moved to AltGr". 
4) Finally activate the new language/keyboard combo it in the taskbar language chooser. (I recall problems with this language selector GUI, though. Maybe it is just buggy or required a logout/login after installing... It works for me now, but if nothing helps, try to enable the old Windows hotkey `LAlt+LShift` to change languages and keyboard layouts, instead of using the taskbar GUI.) 

Once activated, you should now be able to directly type all `RA-*` combinations of characters defined in this keyboard layout, including `ö, ü, ä...`. Of course, `Ö, Ü, Ä` also work when holding shift. However, the default position of the German `ü` on the US intl layout is `RA-y`, which I did not find particularly intuitive. But now you can use your UHK! I defined my own ALT keymap in agent, which simply maps the `u` key to `RA-y`. Likewise for `o=>ö`, `a=>ä` etc. That can be easily and quickly recorded for all keys you need. And because we started with a standard keyboard layout, this UHK level remapping will also work well with existing US intl keyboard layouts found in other operating systems. Btw., Linux already has an US intl layout without dead keys on board, so it is easier than in Windows in this case.

As for Czech characters, I am not sure if all are defined in the US intl layout. Likely, some will be inconveniently hidden behind dead keys, so you would need 2-scancode UHK macros in your ALT keymap that cannot be easily recorded via agent. Then again, you would only have to to this once per language and could distribute it to others via some keymap level export/import option in agent. 
Of course, you could also remap scan codes on the Windows level to any character you want using the keyboard layout creator above. But if you deviate from the standard US intl QWERTY layout and also want to use your UHK with Linux later, you would have to define a custom keyboard layout there as well. I find this more more cumbersome from a design and user perspective, but it might be necessary if US intl is not complete with respect to the Czech characters.

kareltucek, 2020-12-06:
Actually, I use a custom xkb keymap in linux, wich is en-US based plus "special" characters under right alt. I am quite happy with the approach, so this issue is definitely not about fulfilling my needs :-). Portability onto Windows machines is complicated, but fortunately, I do not need to use Windows anymore. Portability onto other linux machines is easy - just copy a file.

> As for Czech characters, I am not sure if all are defined in the US intl layout. 

Me neither, but the above keymap seems to be missing `ů`, and then the group of (internationally more common) `ňžšřčďťě`.  

---------------------------

But generally, from my side this issue is not really about solving international layouts. Its more about:
- people asking a lot for general alt code functionality again and again (esp in the official repository)
- about my own curiosity as to how convenient or inconvenient alt codes would be if used to implement national (or other) layouts
- this keyboard/firmware missing a reasonable feature which would allow mapping such sequences (not just alt codes) easily. I am thinking about some notation like `$tapKeySeq RAS-(u np4 np0)`. The only problem atm. is my laziness to implement the parser for such thing, since implementing it in general (i.e., allowing nested parentheses/modifiers and respecting some sort of rules of modifier continuity), yet stateless and efficient way is a bit tricky.

GrauLab, 2020-12-06:
Thanks. :)
>  I am thinking about some notation like `$tapKeySeq RAS-(u np4 np0)`

I see. I wouldn't go for nested here, though, but keep it simple yet flexible. Using `p` for press, `r` for release and nothing for tap, you could condense my ß above to `$sendKeySeq pLA- np0 np2 np2 np3 rLA-`. This command is essentially just a for loop over existing commands.
If the user defines an impossible sequence like `p-x p-x` or `r-x r-x` or `pLS- LS-x rLS-`, just throw an error.
If you want to go for more user convenience, you could define `$unicodeChar 0223` doing the same as above plus `$suppressMods` and a logic that ensures NumLock is on during the sequence (and restored to its previous state if needed).

As far as "fulfilling needs", I am also happy with respect to international characters, just commenting. ;-) However, I have some other ideas/issues with respect to secondary role triggers by sufficiently overlapping `press->release` time intervals of two keys irrespective of key press order, problems with existing mouse macro commands, and lack of arithmetic speed/acceleration control by macros... Maybe I will create corresponding issues later this year if I find the time, but bye for today.

luteijn, 2020-12-06:
Luckily, Dutch hardly uses 'special' characters, most of them eroded away anyway, and many people generally avoid using the few that are left nowadays, preferring slight misspellings by omitting accents over issues with displaying/printing the proper letters. I can usually get away with standard displayable characters from ASCII 32-126. There is a specific Dutch keyboard layout, but I doubt that anyone uses it. However, sometimes one can't avoid using proper diacritic marks, e.g. when doing schoolwork. 

In the past, when I had to work with windows on the desktop, I used AutoHotkey ( https://www.autohotkey.com/ or see the wikipedia article https://en.wikipedia.org/wiki/AutoHotkey ) to insert 'crazy' characters while just using the non-dead-key US layout, so the quotes, accolades etc. were usable when programming, and when I needed a Euro sign, Ringel-S, or accented character in a text, they were also easily typed. 

Of course, being able to do these things without an agent (as ahk, AutoIT and the like, or even the UHK agent) running on the computer would be preferable, but the way keyboards and their interface work, make some things like this a little hard.

kareltucek, 2020-12-06:
> Using `p` for press, `r` for release and nothing for tap, you could condense my ß above to `$sendKeySeq pLA- np0 np2 np2 np3 rLA-`. 

This actually sounds great! Seems I've been overthinking it :-).

> `$unicodeChar 0223` 

The trouble is that different systems use different conventions, so I would rather not dive into that rabbit hole. 

Still, if we wanted to go super fancy, it should **theoretically** be possible to parse unicode directly from the text input. That is, if Agent's character filters get disabled, and some configuration prerequisites are fulfilled.

> However, I have some other ideas/issues with respect to secondary role triggers by sufficiently overlapping `press->release` time intervals of two keys irrespective of key press order, problems with [...]

I will be happy to read them. And even more happy if you find time to implement them and post a PR :-).

kareltucek, 2022-02-04:
This is now available. Very sorry for taking so long to release it!

closed 2022-02-04 by kareltucek