kareltucek/firmware #56: Question: preserve combined shortcuts when using $holdKeymapLayer

lukaskawerau, 2021-01-31:
I'm on macOS, and I'm trying to implement Emacs "Forward Word" when pressing Option/Alt-F, while preserving other shortcuts like Option-Cmd-F for jumping to the search field.

Getting the right behavior for Alt-F works:

```
$holdKeymapLayer EMC base
```

assigned to Alt on my main Keymap, and `F` set to `Alt-F` on the EMC keymap.
However, I can't get it to work so that when I press Alt-Cmd-F it does not switch to EMC, probably because I don't understand `$ifInterrupted` etc. correctly.

Any help would be super appreciated, and thank you for your fantastic work 🙏

UHKv1
Agent 1.5.8
Firmware 8.10.3 from December 24th

kareltucek, 2021-01-31:
Hi!

If I understand you correctly, then you are interested in `ifShortcut` with `noConsume option`.

For instance:
```
$ifShortcut noConsume 82 final holdKeymapLayer EMC base
$holdKey f
```

(The 82 is obtained by `$resolveNextKeyId` and pressing `f` and identifies the key irrespectively of its mapping.)

lukaskawerau, 2021-01-31:
Awesome, this works!

What would I need to do if I had multiple Emacs shortcuts?

On the EMC keymap I have mapped `f` to `Alt+rightArrow` and `b` to `Alt+leftArrow` - when pressing `Alt` on my main keymap I want hitting either `f` or `b` or to work (again, unless I press `Cmd` on the main keymap). Does that make sense?
Maybe stated differently: switch layer as long as I hold the Alt key on the main keymap and am not pressing any other modifier keys.

kareltucek, 2021-01-31:
Well, right... first a disclaimer - I have a feeling that you are actually trying to mix up functions of a modifier which is actually claimed by two different environments (emacs one, your window manager or window toolkit the other). If it is so, I would recommend not to do such a thing, and to try to reserve a new modifier for your emacs setup. As long as it is two keys, no problem - but once it is half a keyboard... 

Regarding the actual question - I have no special features for such usecase. For small amout of keys/chords, you can simiply create a train of commands. There is many ways to combine them, with diifferent semantics E.g.:

(If  the other modifiers go first)(this would not work for macro-iniduced mods):
```
$ifNotAnyMod final holdKeymapLayer EMC base
$holdKey LA- 
```
(If alt goes first):
```
$ifShortcut noConsume <mod1> final holdKey LA- 
$ifShortcut noConsume <mod2> final holdKey LA- 
...
$holdKeymapLayer EMC base
```
Or simply listing all the chords:
```
$ifShortcut noConsume <f> final holdKeymapLayer EMC base
$ifShortcut noConsume <b> final holdKeymapLayer EMC base
...
$holdKey LA- 
```

kareltucek, 2022-03-05:
Considering this solved due to lack of replies.

closed 2022-03-05 by kareltucek