kareltucek/firmware #57: Mod+D does not hold the Alt modifier on Alt+Tab as default firmware does

Glutnix, 2021-03-26:
Out of the box, the UHK has Mod+D configured to send Alt+Tab. 

On the default firmware (I'm using 8.10.7) holding Mod and tapping the D invokes the Windows app switcher. While holding Mod, the app switcher stays open, so further tapping of D would further switch apps. 

I assume this is because the default firmware 'Holds the Alt' modifier until you either release the Mod key or until another set of modifiers (or none) is required. Regular "alt+tab" works as intended: it is releasing the alt modifier that closes the app switcher.

It also lets Mod+S and Mod+F be used as Ctrl+Alt+Left and Right once the app switcher is open.

However, it seems that using the KT firmware breaks that behaviour. Am I missing something?

I file this against uhk-firmware-8.10.3.kt.1.tar.gz.

kareltucek, 2021-03-26:
Interesting... I have tested it right now against my local binary, and sticky mods on Alt+Tab work just fine. Will give it some time over weekend...

Can you export your userconfig so that I can take a look?

kareltucek, 2021-08-21:
Fixed (in next release). Caused by a typo which broke the ActiveLayerHeld flag. Only one of the two switcher types was affected.