kareltucek/firmware #58: Mapping shift/nonshift scancodes independently doesn't work with secondary shift

WindChimeRan, 2021-05-30:
Mapping shift/nonshift scancodes independently:

$ifShift suppressMods write 4
$ifNotShift write %

With a _primary_ shift pressed, it works fine. But with a _primary_ ( _secondary_ shift, it will output % when holding it.

WindChimeRan, 2021-06-18:
problem solved! The language you designed is super powerful!
```
$ifKeyActive 27 final tapKey 4
$ifKeyActive 85 final tapKey 4
$write %
```

kareltucek, 2021-06-19:
I am happy you have found a workaround to solve your problem, but the reported behaviour is indeed strange and should be fixed or at least investigated. Therefore, please lets leave the ticket open - I definitely do indend to look into it, just have not found the time yet. 

Also sorry for no reply - I definitely did not indent to ignore it like this...

kterui9019, 2021-06-20:
Hello, thanks for providing us with a great tool.
The solution to use `ifKeyActive` will not work `suppressMods`.
```
$ifKeyActive 29 final suppressMods tapKey [
```
expect: `[`
actual : `{`

I specified noOp, as in https://github.com/kareltucek/firmware/issues/41, but it did not work.

kareltucek, 2022-05-07:
Should be fixed in next release.

closed 2022-05-07 by kareltucek