kareltucek/firmware #63: Latest pre-prelease firmware does something strange to your keyboard layout

ewancoder, 2021-08-18:
I've got a Dvorak layout and trying to use the latest pre-release software to tweak my trackpoint sensitivity. Whenever I click "save to keyboard" button - my keys are producing completely different letters. If I switch to qwerty - everything seems ok. If I switch back to Dvorak - I still have wrong weird letters there. Rebooting the keyboard (unplugging - and plugging in again) helps and it fixes the issue so it's not a critical one, I can re-plug the keyboard every time after I edit a macro and save it to the keyboard. However this is something strange. Could you please have a look?

ewancoder, 2021-08-18:
Also one question: is there a way to disable keycluster's trackball as of now?

ewancoder, 2021-08-20:
hm I figured it out. I have a second keymap that is remapping keys from qwerty to dvorak in case you have qwerty set up in the system but need dvorak without adding it in the system itself. this obviously produces wrong keys if you already have dvorak in the OS and you switch to this keymap.

after saving macros configuration to keyboard - keyboard starts to use second (qwerty-to-dvorak) keymap. even though the led screen still shows that it's the first keymap. after using shortcuts to switch between keymaps it works as expected without the need to reboot the keyboard

kareltucek, 2021-08-21:
Thanks for the report! 

I vaguely remember fixing this problem (i.e., accidental layer switch during config reload) when rebasing for UltimateHackingKeyboard/firmware#397 merge. I think it is just not available in current release.

Let's keep this open for now.

kareltucek, 2021-08-21:
> Also one question: is there a way to disable keycluster's trackball as of now?

Yes, there is `none` navigation mode.

ewancoder, 2021-09-05:
Looks like it doesn't reproduce on kt.3 btw.

kareltucek, 2021-09-08:
Yep, think this is solved now.

closed 2021-09-08 by kareltucek