kareltucek/firmware #64: Mouse pointer may perform sudden jumps with the Trackpoint module

llinfeng, 2021-08-21:
With the la

kareltucek, 2021-08-22:
> I tried v8.10.10.kt.2. The same macro no longer works. Pressing the button assigned with the macro won't do anything.

If I remember correctly, kt.1 had  layer-to-navigation-mode mappings messed up, so currently the correct macro should be:

```
$holdLayer mod
$ifNotPlaytime 300 tapKey mouseBtnMiddle
```

> sudden jumps

I remember encountering similar problem when module acceleration exponent was set to too high value. 

Please, flash v8.10.10.kt.3 and try again.

Current defaults for trackpoint are:
```
    { // ModuleId_TrackpointRight
        .speed = 1.0, // min:0.2, opt:1.0, max:5
        .baseSpeed = 0.0, // min: 0, opt = 0.0, max 5
        .acceleration = 0.0, // min:0.0, opt:0.0, max:2.0
        .navigationModes = {
            NavigationMode_Cursor, // Base layer
            NavigationMode_Scroll, // Mod layer
            NavigationMode_Caret, // Fn layer
            NavigationMode_Cursor, // Mouse layer
        }
    },
```

You can tweak the sensitivity using (e.g.):
```
$set module.trackpoint.baseSpeed 0.5
$set module.trackpoint.speed 1
$set module.trackpoint.acceleration 1.0
```

llinfeng, 2021-08-22:
@kareltucek Thanks for pointing out that the `mod` layer should go to the first line. `v8.10.10.kt.3` works as intended.

The tweaking of trackpoint-specific settings is also working as intended. I ended up creating a "Fast" trackpoint macro and a "slow" trackpoint macro, and assigned them to two keys in the Fn layer.

I will stay with the `v8.10.10.kt.3` firmware for now.

Thanks a lot!

closed 2021-08-22 by llinfeng

kareltucek, 2021-08-22:
You are welcome!

llinfeng, 2021-08-23:
By the way, the `NavigationMode_Caret` mode is so cool. Is it documented anywhere else?

kareltucek, 2021-08-23:
There is a "Navigation mode" section in https://github.com/kareltucek/firmware/blob/master/MACRO_DOCS.md (beware, it is a bit ahead of actual state of affairs - new configuration options will not be available until next release - I was heavily refactoring navigation modes yesterday...).

Also, I am sure that Laszlo mentioned it in some blogposts, and official manuals also mention them (e.g., https://ultimatehackingkeyboard.com/manuals/trackball ). 

But implementation details are not documented.

kareltucek, 2021-09-08:
Btw., this seeems to be a case of https://github.com/kareltucek/firmware/issues/64 . 

(Still interested in your findings from running with `acceleration 0`. I am wondering whether acceleration driver could actually be cause for the issue, or maybe just strongly amplifying it.)

llinfeng, 2021-09-08:
@kareltucek Is there a way to "`echo`" the value for the `speed`, the `basespeed` and the `acceleration` for  `module.trackpoint`? I have multiple UHK units and I am currently confused by what each unit has :( And, I did not find anything having to do with "trackpoint" in the UserConfiguration.json file (exported from Device Configuration).

My current take is that setting `acceleration 0` and flashing with the new firmware removed 99% of those surprising jumps. Still, I need to force the same "trackpoint profile" on all my keyboards to reach a conclusive answer about the remaining 1% jumps. Will update here in a week.

llinfeng, 2021-09-08:
Wait, it seems that the default setting is having `acceleration 0` for track points (per your first reply in this issue)? Then, I still have a few fixed-length jumps that are either upward or rightward. 

@kareltucek Please advise if the following test helps. I am assuming flashing the firmware shall reset trackpoint settings to default, which has `0` for accelaration?
* On all UHK+Trackpoint sets, flash the v8.10.10.kt.3 firmware afresh, and stay away from tweaking the sensitivity settings.

kareltucek, 2021-09-08:
> Is there a way to "echo" the value for the speed, the basespeed and the acceleration for module.trackpoint? 

Not directly. 

But for instance throwing something like `setLedTxt 500 ABC` into the configuration macro could help to keep track "manually".

> And, I did not find anything having to do with "trackpoint" in the UserConfiguration.json file (exported from Device Configuration).

Indeed. Almost all values are "hardcoded" in the firmware, and the command-oriented settings is just a work-around to be able to set them without having to alter the stock serialization in any way. 

> @kareltucek Please advise if the following test helps. I am assuming flashing the firmware shall reset trackpoint settings to default, which has `0` for accelaration?
> 
>  * On all UHK+Trackpoint sets, flash the v8.10.10.kt.3 firmware afresh, and stay away from tweaking the sensitivity settings.

No. Setting them explicitly to `acceleration 0` should be equivalent to staying away.

---------------------------------

> Then, I still have a few fixed-length jumps that are either upward or rightward.

Let's say that I provide you a stock firmware that has acceleration driver completely removed. Would you be willing to give it a try for a few days and confirm whether the issue persists?

llinfeng, 2021-09-09:
@kareltucek Sure, I can try an alternative firmware for a few days. What should I expect for [your text actions fix](https://github.com/UltimateHackingKeyboard/agent/issues/1572#issuecomment-895475871) using the firmware in this repo? I am fine living without it, but will surely miss it :)

----
 
I haven't yet created a new macro to force `acceleration 0` for the trackpoint. Instead, I may have caught one jump after flashing the first trackpoint module I received with the `v8.10.10.kt.3` firmware. Can this be a hardware related issue? The buttons are a lot louder on the first trackpoint module, compared to the other three trackpoint modules I got over subsequent orders via the UHK store.

I will add the following macro to a key to force the following settings, and wait for our altered firmware to drop.
```
$set module.trackpoint.baseSpeed 0.1
$set module.trackpoint.speed 0.8
$set module.trackpoint.acceleration 0
```

kareltucek, 2021-09-09:
> What should I expect for using the firmware in this repo? 

Sorry, I don't understand the question. Would you rephrase?

---

The no-acceleration vanilla firmware is here
[uhk-firmware-8.10.10-no_accel.tar.gz](https://github.com/kareltucek/firmware/files/7138375/uhk-firmware-8.10.10-no_accel.tar.gz)

llinfeng, 2021-09-11:
@kareltucek I haven't observed a single "jump" with the `no_accel` firmware. This is from two pairs of UHK+trackpoint modules.

I will rewind to the previous test where I set `acceleration 0` manually. One question: how often should I call/activate the macro that sets the following values after flashing the UHK+trackpoint with [v8.10.10.kt.3](https://github.com/kareltucek/firmware/releases/tag/v8.10.10.kt.3)? Does it suffice to call the macro once only?
```
$set module.trackpoint.baseSpeed 0.1
$set module.trackpoint.speed 0.8
$set module.trackpoint.acceleration 0
```

kareltucek, 2021-09-11:
> I haven't observed a single "jump" with the no_accel firmware. 

Interesting. I guess I will ask in UHK repo just to confirm. 

> I will rewind to the previous test where I set acceleration 0 manually. 

I guess there is no point in further testing that setup.

> One question: how often should I call/activate the macro that sets the following values after flashing the UHK+trackpoint with v8.10.10.kt.3? Does it suffice to call the macro once only?

Since the values are similar to defaults, there is no need to call the macro.

Otherwise, after powercycling UHK, and after flashing firmware.

llinfeng, 2021-09-11:
Well, we may have a setback with testing the `no_accel` firmware - I just saw a jump when I was trying to move the cursor to the left. It jumped by some 400 pixels to the right, when I first touched the trackpoint. (If I recall it correctly, I was resting my index finger on the trackpoint doing some continuous mouse movement. The jump happened when I was about to reach the restart button below. While doing the fine adjustment to click on the restart button, the cursor jump horizontally to the right, covering roughly 1/3 of the width of my monitor, which is 1200 pixels.)

[image]

With the `no_accel` firmware, though, the frequency for the jumps indeed get a lot lower. This is the third day of testing, where I may have logged some 10+ hours on to pairs of UHK+trackpoint pairs. The jump I described came from a "new" trackpoint model ordered directly from the UHK store.