kareltucek/firmware #68: Prefix keypresses with tapKey in secondary role

ronyclau, 2021-09-17:
I am trying to map a key on my key cluster to send a single `C-w` in primary role, and prefix all key presses with `C-w` while in secondary role.

I can make the first key press after holding the key work, however, all subsequent key presses while the cluster key is still depressed only send out the key itself.

Expected:
1. Hold the cluster key
2. Press `l` (Send `C-w` then `l`)
3. Press `j` (Send `C-w` then `j`)
4. Release the cluster key

Actual:
1. Hold the cluster key
2. Press `l` (Send `C-w` then `l`)
3. Press `j` (Send `j`)
4. Release the cluster key

I've tried using a loop with `goTo` and `ifInterrupted`, however, as the docs stated, the `ifInterrupted` condition is evaluated only once and thus didn't work as I intended.

Is there any way to implement this using the current macro system? Thanks!

kareltucek, 2021-09-17:
Condition is evaluated once per line execution. I.e., if you execute for instance `ifShift delayUntilRelease`, the delay will end as expected on macro key release, and not when a shift is released. Your problem with ifInterrupted is just that the flag once set, it is not resetted.

For the same purpose, I am using following sequence:

```
begin: postponeKeys ifNotPending 1 ifNotReleased goTo @0
postponeKeys ifReleased final ifNotInterrupted ifNotPlaytime 300 tapKeySeq @ @
postponeKeys ifPending 1 tapKey @
postponeKeys setReg 0 %0
ifPending 1 goTo @0
ifKeyActive #0 goTo @0
goTo begin
```

But i must admit, that I don't like it - very complex for a relatively sane and common usecase. Any way to introduce some new syntax (syntatcti suggar, command, whatever) that would simplify the thing?

ronyclau, 2021-09-20:
Thanks for the reply and it's working perfectly, though I do share the thought that this macro is looking far too convoluted for a simple use case like this.

Perhaps we can have a condition like `whileHeldAndInterrupted` or `whileNotReleasedAndInterrupted` (I am bad at naming 🤦) that abstracts away the logic:
```
$ifPrimary final tapKey C-w
$whileHeldAndInterrupted tapKey C-w
```

(I thought of `whileHeld` but it sounds like the action would be invoked in a tight loop while the key is held down. `ifInterruptedWhileHeld` sounds better, but doesn't conform to the convention that all `if*` conditions are only evaluated once.)

Or perhaps we may solve this another way round by introducing a new command that resets the states of all evaluated conditions, and thus enables re-evaluation next time a condition is encountered? So we may write something like this:

```
$ifPrimary final tapKey C-w
$ifInterrupted tapKey C-w
$ifReleased break
$resetConditions
$goTo @-3
```

Anyhow, thanks so much for your hard work on this firmware!

kareltucek, 2021-09-20:
Doesn't work very well :-(. The problem itself is much more about operating the postponer queue than about simple `ifInterrupted` logic. With `ifInterrupted`-induced activation alone, you should expect your prefix action to happen after the other keypress rather than before. (The `ifPrimary` probably saves the situation, since it forces postponing mode.) Yet then, there is another issue - it is necessary to wait for key release before resetting the conditions, because otherwise it will get stuck. Also, ifPrimary has its own pack of side-effects, since it definitely wasn't intended for this purpose. 

I guess we will have to wait till more usecases pop up that will allow us to determine correct way to abstract this...

closed 2021-09-20 by kareltucek