kareltucek/firmware #74: Parameterised auto shift?

jedahu, 2021-11-30:
Is there any way to make the solution to #70 parameterised on the pressed key?

Something like:
```
delayUntilReleaseMax 350
ifPlaytime 350 final tapKey S-<pressed-key>
tapKey <pressed-key>
```

Which could then be bound to multiple keys.

kareltucek, 2021-12-01:
Theoretically following, assuming that you have up-to-date firmware (e.g., https://github.com/UltimateHackingKeyboard/firmware/files/7464023/uhk-firmware-8.10.10-tp_hunt_v11.tar.gz).

```
delayUntilReleaseMax 350
postponeKeys activateKeyPostponed atLayer mod #key
ifPlaytime 350 final pressKey S-
delayUntil 20
```

This should enqueue the same key's mod layer action for execution, and then either press shift or not. If things go well, the enqueued key will get tapped while the shift key is still held.

But I do not like the code - it's tricky and I am not even sure that it should by definition work reliably. I will leave this ticket open for later reevaluation of the problem.

jedahu, 2021-12-01:
Thanks. I'll try it out!

jedahu, 2021-12-01:
I have the correct firmware. The hold-delay works, but both tap and hold write '7', regardless of which key I bind the macro to.

[image]

kareltucek, 2022-05-07:
Sorry for so late reply!

Testing following does the trick as long as just one key is pressed at the time.

```
postponeKeys delayUntilReleaseMax 350
ifPlaytime 350 pressKey S-
activateKeyPostponed atLayer mod #key
delayUntil 50
```

I see that it is not sufficient - the usecase would need a dedicated support. I will leave the ticket open, but I do not plan to pursue it at the moment.

jedahu, 2022-05-09:
No worries. Thanks for the follow up.

kareltucek, 2022-09-30:
-> https://github.com/UltimateHackingKeyboard/firmware/issues/525#issuecomment-1263492412