kareltucek/firmware #75: Odd behaviour with modifiers

rafaelpirolla, 2021-12-15:
I'm using Karabiner Elements EventViewer to check this behaviour.
From base use a macro with "$toggleLayer fn" to switch to another layer.
Use a key with two modifiers and the events are reported as: 
type:down            HID usage: 7,227     name:{"key_code":"left_command"}                             misc:flags left_command
type:down            HID usage: 7,226     name:{"key_code":"left_option"}                              misc:flags left_command,left_option
type:down            HID usage: 7,79      name:{"key_code":"right_arrow"}                              misc:flags left_command,left_option
type:up              HID usage: 7,79      name:{"key_code":"right_arrow"}                              misc:flags left_command,left_option

And then only when I switch back to another layer with $toggleLayer is that I see the up events:
type:up              HID usage: 7,227     name:{"key_code":"left_command"}                             misc:flags left_option
type:up              HID usage: 7,226     name:{"key_code":"left_option"}                              misc:

It could be that some other key would trigger the up events but I didn't test too many options. I think this would suffice to trace a bug or misconfiguration on my part.

kareltucek, 2021-12-15:
Looks like an official feature called "sticky modifiers". Its purpose is to make shortcuts like "alt+tab" work when they are bound in non-base layers. However, it should fire only in "held" layers.

I will look into it when revising the layer switching code. Unless you have such bindings, quick solution is to `set stickyMods never`, (e.g., in a macro named `$onInit` in order to make the config persistent).

rafaelpirolla, 2021-12-16:
It would work indeed, I've tested. But I'm using toggleLayer to switch into layers.

kareltucek, 2021-12-16:
> It would work indeed, I've tested. But I'm using toggleLayer to switch into layers.

I don't understand. Do you mean that disabling sticky mods does not help when you use `toggleLayer` command?

kareltucek, 2022-05-07:
I have retested the scenarios and stickies seem to work as expected. 

Closing, but feel free to followup if anything is unclear or does not work as expected.

closed 2022-05-07 by kareltucek