kareltucek/firmware #78: Runtime macros

tunney, 2021-12-23:
Perhaps user error but cannot seem to get runtime macros to work:

Even the example:
$ifShift recordMacro A
$ifNotShift playMacro A

This is my configuration:

[image]

Hitting FN+Shift+8 results in the triangle lighting up
Then type some characters, say 'qwerty'
Hitting FN+Shift+8 results in the triangle lighting up

Then my expectation was hitting FN+8 would result in 'qwerty' being printed.

Suggestions and thoughts?

kareltucek, 2021-12-23:
When exactly does the triangle go dark?

In your case, the second FN+Shift+8 hit apparently *starts* recording again, which deletes the already-recorded macro, so the FN+8 cannot really produce any result...

tunney, 2021-12-23:
It does dark as soon as I release FN+SHIFT+8

tunney, 2021-12-23:
Actually it goes dark before I release

kareltucek, 2021-12-23:
This is weird. Please try again with [uhk-firmware-8.10.12-prerelease.tar.gz](https://github.com/kareltucek/firmware/files/7768955/uhk-firmware-8.10.12-prerelease.tar.gz).

If there is still a problem, then please attach your UserConfig.

tunney, 2021-12-23:
Same behaviour on 8.10.12-prerelease

Does it make a difference if I am on a V2?

tunney, 2021-12-23:
This time with user config
[dtunney_2022.12.23_11.25.json.txt](https://github.com/kareltucek/firmware/files/7769021/dtunney_2022.12.23_11.25.json.txt)

kareltucek, 2021-12-23:
Should not matter, except that the hotswaps might be more prone to problems than soldered switches. Actually I have just tested with a V2 prototype and your userconfig, and it works just fine.

Does your 8 key function reliably? (Please try it with another key.)

tunney, 2021-12-23:
I tried with the 'X' key and same.

I have two V2s (thought I would be returning to an office when I ordered) and bought behave the same.

tunney, 2021-12-23:
Also if i do a "normal" macro that just prints text the LEDs display ERR

kareltucek, 2021-12-23:
> Also if i do a "normal" macro that just prints text the LEDs display ERR on the prerelease firmware

Sorry about that - I have made the firmware build parse any macro as a command action, since up-to-date firmware is already migrated to the command action (otherwise, you would need custom builded agent to test with the new firmware).

----

Regarding the original problem:

- Does the ERR on your display go away on its own, or does it last until either a keymap is switched or the keyboard is powercycled? (Similarly, if you hit FN+Shift+8 while ERR is on your display, does it go away with the triangle going dark?)

- can you try with different computer? 

- are you using any UHK modules?

- if you add `$tapKey x` at the beginning of the macro, save config, powercycle your UHK, focus a text editor, and then tap FN+Shift+8, does it produce `X` or `XX`?

tunney, 2021-12-23:
> Regarding the original problem:
> 
> * Does the ERR on your display go away on its own, or does it last until either a keymap is switched or the keyboard is powercycled? (Similarly, if you hit FN+Shift+8 while ERR is on your display, does it go away with the triangle going dark?)
Stays until power cycled, or keymap switch
FN+Shift +8 does not remove the ERR but the triangle comes on and goes off

ERR only comes with with text only
> * can you try with different computer?
Something to this one.
It works on a crappy Windows 10 machine, I mean a terrible laptop
On a macbookPro it does not..... (this was the last step I tried)
What does this mean?

[image]

> * are you using any UHK modules?
Key cluster and trackball, removed, unplugged/replugged and does the same

> * if you add `$tapKey x` at the beginning of the macro, save config, powercycle your UHK, focus a text editor, and then tap FN+Shift+8, does it produce `X` or `XX`?
It displays an 'X'
What should it do?

tunney, 2021-12-23:
Tried on a 2017 macbook air and it didn't work
And then tried on a Dell Winwdows laptop and it did work

Seems to be Mac specific?

kareltucek, 2021-12-23:
> What should it do?

Just one X is fine.

(Sorry about the chaotic edit - I didn't realize I had sent the long version before pruning it down.)

----

> What does this mean?

Maybe mac-specific, maybe motherboard-specific. 

There has been a fix which might be related in stock repo, so I will first merge that fix and build a new firmware to test with. I don't have time to deal with it right now though, so bear with me :).

tunney, 2021-12-23:
> Just one X is fine.
>
> (Sorry about the chaotic edit - I didn't realize I had sent the long version before pruning it down.)
>
>
> Maybe mac-specific, maybe motherboard-specific.
>
> There has been a fix [...]

I am just glad I wasn't being a total idiot and doing something stupid!

kareltucek, 2021-12-24:
Here we go - [uhk-firmware-8.10.12-prerelease.tar.gz](https://github.com/kareltucek/firmware/files/7774250/uhk-firmware-8.10.12-prerelease.tar.gz) . 

If this firmware does not fix the problem, next step is to get in touch with Laszlo (either customer service email, or the UHK github repo), since this issue does not look like a bug in my code. Before doing so, it might be a good idea to gather some more evidence about the problem. E.g., to check system logs, or a way to reproduce the problem with regular UHK features (e.g., if you switch to non-default keymap, does your UHK remain there?). 

(I suspect something like UHK being disconnected and reconnected in regular intervals, and therefore loosing stateful information.)

tunney, 2021-12-24:
8.10.12-prerelease2 fixes this issue. Runtime macro are working on my mac. Little triangle stays on.
A nice little Christmas present - thank you.

I am not sure on the ticket workflow here - shall I just close or is there some other process required?

kareltucek, 2021-12-24:
Glad to hear it!

closed 2021-12-24 by kareltucek