kareltucek/firmware #81: Using space character in text

soraxas, 2021-12-24:
Hi! What would be the best way to input space characters in the macro engine?
For example, to display " HI" as the led text.

kareltucek, 2021-12-24:
Currently there is no way to do that. (Let's call it technical debt.)

At some point, the language definition should be refactored so that strings are regular unambiguously-recognizable tokens (some type of quote notation). This loosely relates to https://github.com/UltimateHackingKeyboard/agent/issues/1630 and its plan to provide syntax-aware editor features - I assume that the refactor may be either necessary, or desirable in order to make those features work.

(Also, during such refactor, other features may be considered - like scoping or expression parser.)

soraxas, 2021-12-25:
I think it's amazing how this macro engine had implemented such a workable token parsing engine for the macro engine. But indeed it is always cumbersome to reinvent string token parsing. Perhaps we should leverage existing scripting engines (or at least get inspiration on how they are implemented) to avoid re-inventing the wheels?

For example, [this lisp interpreter in C](https://github.com/justinmeiners/lisp-interpreter) looks interesting and it directly follows the operator / args expression that we currently have. The brackets also helps scoping the expression (and chaining actions)

kareltucek, 2021-12-25:
>  (or at least get inspiration on how they are implemented) to avoid re-inventing the wheels

This is totally agreed.

> Perhaps we should leverage existing scripting engines

Please note that we are very tight on RAM. 

UHK has 128 kb of RAM out of which 2x 32 kb are taken by configuration. It might be possible to free one of these regions by some clever refactor. The 64 kb are almost depleted - there are some 3kb of reserve in buffers - and we will probably need to free some of it for custom RGB backlight. I am not sure how good idea it would be to introduce malloc into the codebase - so far, everything is statically allocated, and the macro engine is designed so that it is almost stateless.

> For example, this lisp interpreter in C looks interesting and it directly follows the operator / args expression that we currently have. The brackets also helps scoping the expression (and chaining actions)

I will give it more thought another time. However, if you are eager to dig deep into it, feel free to.

kareltucek, 2022-05-09:
I have looked briefly into the interpreter, and realized that I am not sure how to maintain stateful information if the interpreter is employed.

It seems to me, that either a lot of the logic of "modifier commands" would either have to be re-implemented in lisp and consequently loaded as a lisp library, or a lot of state would have to be maintained outside of the lisp interpreter itself, including scoping.

As a result, it seems like employing the interpreter would mean having to write and maintain an engine which would actually execute as a "Cartesian product" of the current macro engine *and* the lisp interpreter. I.e., the solution would have to switch between C runtime and lisp interpretation modes in a very intricate and delicate manner.

I have a feeling that lisp, as a pure functional language is well suited for computating results, but badly suited to do effectful processing. In this sense, current macro language is 100% procedura language, which does 100% of its work by performing effects, rather than computing values in a functional sense.

kareltucek, 2022-05-09:
(This is assuming that we would actually want UHK to run a full fledged lisp interpretter.)

kareltucek, 2023-09-01:
Is/will be solved by https://github.com/UltimateHackingKeyboard/firmware/pull/674

closed 2023-09-01 by kareltucek