kareltucek/firmware #84: Send release key signal for keys not activated by macro

soraxas, 2021-12-25:
Hi sorry to bother you again! 
Do you know if it's possible to send a key-up event for keys presses that is not emitted by the same macro?

In the docs, `releaseKey` it stated that
> ... This does not affect scancodes emited by different keyboard actions

which make sense. My use case is that I want to bind `D` to vim's cut to end-of-line, which I had set it up like
```
ifShift tapKeySeq S-end C-x
```
However, since the macro key `D` requires me to hold shift, the cut action becomes `ctrl`+`shift`+`x` which does not work (only ends up highlighting the line). I've tried `ifShift tapKeySeq S-end S- C-x` to add an empty shift press but it still doesn't work. Thanks for any advice on this!

kareltucek, 2021-12-25:
> Hi sorry to bother you again!

No trouble. I will try to provide you with information in real time, but leave feature requests / pull requests for another time...

> Do you know if it's possible to send a key-up event for keys presses that is not emitted by the same macro?

Directly and generally, it is not possible - resulting usb report state is sum of us usb_repot_updater's usb repots and macro state reports - each macro slot/state has its own set of reports, which it uses to process tap/hold/release/press events. These are alive for as long as the macro is active, and then freed.

For more advanced features (for stateful vim-like things), you can use registers (search for `setReg`) to signal between macros.

Regarding the shift issue, there is `suppressMods` modifier, which should suppress regular scancode action modifiers. (This will not work if the shift is triggered/produced by a macro.) (Usually, folks want to bind shift/nonshift scancodes independently - yours is just a more advanced usecase of the same thing.)

soraxas, 2021-12-26:
> No trouble. I will try to provide you with information in real time, but leave feature requests / pull requests for another time...

You are too kind!

> Regarding the shift issue, there is suppressMods modifier, which should suppress regular scancode action modifiers. 

Thanks for the pointer!

Actually I've lied: previously, I had followed the guide and mapped my shift key to double tap to caps
```
holdKey leftShift
ifDoubletap tapKey capsLock
```
... which, of course it doesn't work with `ifShift`, so my setup was actually
```
ifKeyActive 85 tapKeySeq S-end C-x
```
*(where keyid 85 is leftShift)*.

`supressMods` doesn't work with mods-by-macro. But I don't care too much about double-shift-to-caps so I've switch my shift back to a normal key, and `suppressMods` works wonder. Thanks :)

---------------------

I guess `suppressMods` doesn't work across macro because each macro holds its own bitfield of USB report (to avoid pollution across macro?), and therefore, a releaseKey event from *macroA* has no effect on *macroB* when they are bitwise-AND?

kareltucek, 2021-12-26:
> I guess `suppressMods` doesn't work across macro because each macro holds its own bitfield of USB report (to avoid pollution across macro?), and therefore, a releaseKey event from _macroA_ has no effect on _macroB_ when they are bitwise-AND?

Much more prosaic reason - because in context of a macro, you usually want to override non-macro behaviour, and therefore actually want to apply macro-induced-shifts irrespectively of applied `suppressMods`.

E.g., in case of `ifShift suppressMods tapKey C-w`, you want that control present.

Anyway, this is another small technical debt - if you can come up with sensible specification which will make things work better and not break too many current usecases, then a PR is welcome.

kareltucek, 2022-05-07:
> Anyway, this is another small technical debt - if you can come up with sensible specification which will make things work better and not break too many current usecases, then a PR is welcome.

I have refactored modifier handling some time ago into a concept of Input, Output and Sticky modifiers, so the behaviour should now be clearly defined, and controllable, so I consider it solved. Let me know if the manual is unclear or if you have any other remarks!

closed 2022-05-07 by kareltucek