kareltucek/firmware #86: is there a way to clear modifiers set inside a macro?

herrsimon, 2022-01-06:
Hello,

first of all, thanks for the great macro engine!

I currently have left ctrl set up with a space cadet like behaviour as follows:

```
$holdKey leftControl
$ifInterrupted break
$ifPlaytime 200 break
$tapKey graveAccentAndTilde
```

Now I would like to add the following behaviour: If ctrl is held down and some specific key, say `a`, is pressed, the scancode should be sent WITHOUT the ctrl modifier set. However, neither 'ifCtrl', nor `suppressMods` works here in a macro for the `a` key, as ctrl has been set inside a macro. Is there some other way to obtain this behaviour?

Thanks in advance!

kareltucek, 2022-01-07:
Try to replace `holdKey leftControl` by `$ifNotShortcut 79 holdKey leftControl`. It will delay press of the control, so it might interfere with external mouse, and it may need some tweeking so that it does not break rest of the keyboard.

(This is just a workaround proposal, but it is the best I can do now. Generally, this is a known problem and the modifier handling needs to be redesigned in order to allow usecases like yours.)

herrsimon, 2022-01-07:
Actually, I already have mouse issues with some unidentified macro (the two buttons on the thumb cluster stop working regularly and only replugging makes them work again), but I will still try your workaround. Thanks for the quick reply and keep up the excellent work!

kareltucek, 2022-01-07:
> Actually, I already have mouse issues with some unidentified macro (the two buttons on the thumb cluster stop working regularly and only replugging makes them work again)

This is strange. If you can figure out the cause, please let me know.

herrsimon, 2022-01-08:
I just tried for half an hour to make the buttons stop working but of course it didn't happen. Basically, I only have space cadet like macros on caps lock and all alt, control, shift and win keys, so nothing fancy. Maybe it also has to do with the fact that I load a custom xkb layout via `xkbcom` (instead of placing the files into the X11 config dir), although I don't have an explanation how this could happen. In any case, I will keep investigating. Have a nice weekend!

herrsimon, 2022-01-08:
I meant `xkbcomp` of course.

kareltucek, 2022-05-07:
> Actually, I already have mouse issues with some unidentified macro (the two buttons on the thumb cluster stop working regularly and only replugging makes them work again)

Reminds me of a power issue which was reported a few days ago...

Anyway, closing due to lack of further followups.

closed 2022-05-07 by kareltucek