kareltucek/firmware #88: lowecase letters stopped working on LED display

kareltucek, 2022-01-18:


soraxas, 2022-01-23:
Do they suppose to work? their values (segment map) in the source seems to be null:
https://github.com/kareltucek/firmware/blob/common_trunk_upstream/right/src/led_display.c#L81=L106
```c
    0b00000000000000, // a
    0b00000000000000, // b
    0b00000000000000, // c
    0b00000000000000, // d
    0b00000000000000, // e
    0b00000000000000, // f
    0b00000000000000, // g
    0b00000000000000, // h
    0b00000000000000, // i
    0b00000000000000, // j
    0b00000000000000, // k
    0b00000000000000, // l
    0b00000000000000, // m
    0b00000000000000, // n
    0b00000000000000, // o
    0b00000000000000, // p
    0b00000000000000, // q
    0b00000000000000, // r
    0b00000000000000, // s
    0b00000000000000, // t
    0b00000000000000, // u
    0b00000000000000, // v
    0b00000000000000, // w
    0b00000000000000, // x
    0b00000000000000, // y
    0b00000000000000, // z
```

kareltucek, 2022-01-23:
Well, I definitely do expect them to work. The change which made them stop working was (unintentionally) merged from `uhk/master`, which in the past diverged significantly. I am quite confident that current situation is not result of a deliberate decision.

Either way, I can easily fix it.

kareltucek, 2022-02-04:
Done.

closed 2022-02-04 by kareltucek

soraxas, 2022-02-05:
Would it be better to support lower case letters like so?
https://www.myfonts.com/fonts/matthias-luh/14-segment-led-display/

(you can test typing the font on the website)

kareltucek, 2022-02-05:
I don't have a strong opinion on that. I just want to get a readable output rather than empty LED display.

- I fear worse readability.

- Regarding those specific fonts, I really don't want to have to deal with EULAs and licensing.

- I don't find it worthwhile.

(If you want to dig into this further, I suggest doing so in the official repo, as Laszlo might have a preference of his own.)

soraxas, 2022-02-06:
Yeah make sense, I don't have any strong opinion, but it's just that after seeing the fix I was wondering if it's better to add to variability rather than duplicating functionalities. The readability issue is a non-issue as the upper-case equivalent remains to be an option (and they corresponds closer to what the user wants anyway).

Perhaps the more generic approach is to allow arbitrary bitfield (of the 14 segments) to be controllable via macro (perhaps with the individual `AlphanumericSegmentsBrightness` to be controllable per segment as well)