kareltucek/firmware #94: Scrolling and Trackball can't work at the same time.

mrpond, 2022-02-23:
Install the latest Agent:
Agent version: 1.5.17
Use Agent to update to the latest firmware:
Right keyboard half firmware: 8.10.12
Left keyboard half firmware: 8.10.12
Key cluster firmware: 8.10.12
Trackball firmware: 8.10.123.

refer to this issue:
https://github.com/UltimateHackingKeyboard/firmware/issues/452

after apply set module.trackball.navigationMode.mod cursor

While scrolling with left module I can't move UHK's trackball cursor.

kareltucek, 2022-02-23:
In which scenarios you need to scroll and move cursor at the same time?

kareltucek, 2022-02-23:
But yes, this is a relevant ticket. Only one navigation mode can currently be active at a time.

mrpond, 2022-02-23:
> In which scenarios you need to scroll and move cursor at the same time?

a lot

mondalaci, 2022-02-28:
We've just received a report of this issue, so I'm attaching a 9.0.0 pre-release including this fix: 
[uhk-firmware-9.0.0-df83a18.tar.gz](https://github.com/kareltucek/firmware/files/8153164/uhk-firmware-9.0.0-df83a18.tar.gz)

mrpond, 2022-03-05:
> We've just received a report of this issue, so I'm attaching a 9.0.0 pre-release including this fix: [uhk-firmware-9.0.0-df83a18.tar.gz](https://github.com/kareltucek/firmware/files/8153164/uhk-firmware-9.0.0-df83a18.tar.gz)

I go test it for you.. ^^

mondalaci, 2022-03-05:
@kareltucek Feel free to close this issue then. I don't have permission, so I can't :)

kareltucek, 2022-03-05:
@mondalaci I know, I intended to close it when I get to publish a release.

kareltucek, 2022-03-05:
Done

closed 2022-03-05 by kareltucek