kareltucek/firmware #97: Q: map Mod+Shift to different macro?

laur89, 2022-04-12:
Hi, this is more of a question than an issue.
First raised it on [one of UHK's threads](https://ultimatehackingkeyboard.com/blog/2018/06/23/how-can-i-type-accented-characters-with-my-uhk?unapproved=135432&moderation-hash=15a9918a5c6aff6b5170ff1873d81ed0#comment-135432):
```
Is there a way to trick keyboard to send scancode for upper-case unicode character when Shift modifier is held down?
I'm running QWERTY PC map on linux, US layout.

Trying to map some keys to type öäü via Mod layer. It works fine after defining alt-code macros. But problem is you 
can either define upper _or_ lower case combination.
Is it possible to send different macro when Mod+Shift is held down?
```

To which László replied your fork might fit the bill. Can you confirm that's the case?

kareltucek, 2022-04-12:
Yes, it is the case. 

Please search https://github.com/kareltucek/firmware/blob/master/MACRO_DOCS.md for `ifShift`, I am sure you will hit something in the example section.

You will just want to replace the simple write or tapKey command with either a `tapKeySeq` or with `exec MACRONAME`.

kareltucek, 2022-05-07:
Closing due to lack of replies...

closed 2022-05-07 by kareltucek