kareltucek/firmware #98: Scroll and click with the same Trackpoint button - with UHK 2

llinfeng, 2022-04-27:
With UHK 1, the following macro is working as intended when assigned to the right-button on the Trackpoint module. A single-click on the button issues a middle click, and holding the button gets me to scroll in four directions with the Trackpoint. [Here are a few [related](https://github.com/kareltucek/firmware/issues/64) [issues](https://github.com/UltimateHackingKeyboard/firmware/issues/383) where we got this ideal behavior for UHK1.]
```
$holdLayer mouse
$ifNotPlaytime 2000 tapKey mouseBtnMiddle
```

With UHK 2 (flashed with `uhk-firmware-v8.10.12.kt.2.tar`) from [this release](https://github.com//kareltucek/firmware/releases/tag/v8.10.12.kt.2), the scrolling part fails to work. When pressed, a single mid-click is sent as intended. When held, moving the Trackpoint shall move the mouse cursor still freely, and a mid-click is issued at the original spot when the button is released.

It would be nice to restore how the right-button on the Trackpoint module behaves when attached to a UHK2 keyboard. Before the fix, "thanks" to Lenovo that places a Fn key at the far-left corner of their keyboard, assigning that key as a `Mod` key in keymap gets me to scroll around with the Trackpoint with ease.

kareltucek, 2022-04-28:
I failed to reproduce the problem.

So when you hold the key, does the `holdLayer` switch itself work as intended? (I.e., does the "mouse" led light up for as long as you hold the key?)

Also, if possible, please attach your UserConfig.

llinfeng, 2022-04-28:
I was wrong about where the mid-click is sent upon release - upon releasing the key on the Trackpoint, a mid-click is completed at the spot of the mouse cursor.

> So when you hold the key, does the holdLayer switch itself work as intended? (I.e., does the "mouse" led light up for as long as you hold the key?)

The "mouse" LED light did light up when I hold down the right button on the Trackpoint module. It goes off when I release the button.

Please find my UserConfig [here](https://www.dropbox.com/s/4z8bop7ma10w4m1/UserConfiguration_2022-04-28_UHK2.json?dl=0). This is the version I used when reporting this issue yesterday.

----

@kareltucek Thanks for asking about the `mouse` light! This actually reminds me of a previous conversation, where we were mistaking the `mod` and the `mouse` mode. Actually, with `uhk-firmware-v8.10.12.kt.2`, the scrolling action is working fine under `mod` mode. Thus, updating the macro by replacing "mouse" to be "mod" helped achieve the ideal behavior. Here are the two lines I keep in the macro.
```
$holdLayer mod
$ifNotPlaytime 1000 tapKey mouseBtnMiddle
```

And, the two lines in the macro look like the following when applied to the UHK V2 keyboard.
[image]

[image]

closed 2022-04-28 by llinfeng