kareltucek/firmware #100: How to use Windows Key in tabKeySq?

provinzio, 2022-06-28:
Hey,

I'd like to include a Windows Shortcut into a macro chain.
Unfortunatly I can't find the keyid for a windows key (already looked in `macro_shortcut_parser.c` and the documentation).

Would love a kick in the right direction.

kareltucek, 2022-06-28:
It is the "Gui" modifier. So if you want the key alone, "LG-" should work.

provinzio, 2022-06-29:
Thanks for the heads up.

I am still having difficulties with the command. My OS layout is QWERTZ.

Using `$tabKey LG-` gives the error `testÖ 0Ö 0Ö unrecogniyed commandÖ tabKez LGß`. So I thought, I have to mimick a en-US layout, `y` turns to `z` but which key do I use for the `-`? At the position of the en-US `-` has QWERTZ `ß` which can not be entered in the text macro field.

provinzio, 2022-06-29:
For my first test, I wanted to implement `Win+L`.

Running the following does work.
```
$pressKey leftGui
$tapKey l
$releaseKey leftGui
```

I also tried to implement this with `tapKeySeq`, but as far as I found out, I cannot use the shortcuts e.g. `LG-` because, I cannot enter the scancode for enUS `-` into the agent - which is `ß` on my layout. Pressing the `ß` on my keyboard does nothing when trying to enter the text macro.

So I cannot use `tapKeySeq`? Or is there some other way?

kareltucek, 2022-06-29:
> For my first test, I wanted to implement `Win+L`.

You may as well use `tapKeySeq LG-l`, or `tapKey LG-l` for that matter, since the `LG-l` is still just a simple shortcut - consisting of a modifier and a key. (The windows/gui key is a *modifier* (just as a shift or control).)

 (The left hand side of `LG-l` stands for modifier mask and means Left Gui.)

You could also rewrite the above explicitly as `tapKeySeq pLG- l rLG-`, meaning `press left gui, tap l, release left gui`.

> I also tried to implement this with `tapKeySeq`, 

Meaning you are trying to implement the `Win+L`, while facing the problems with the `ß` key? 

Commands are consisting of *characters*, not scancodes, so you are really supposed to just paste `tapKeySeq LG-l` into the field. (The reason why error log types out `ß` is that... uhk types out actual `-` scancode, and it is your OS which translates it to `ß` because of your language map.)

> but as far as I found out, I cannot use the shortcuts e.g. `LG-` because, I cannot enter the scancode for enUS `-` into the agent - which is `ß` on my layout. Pressing the `ß` on my keyboard does nothing when trying to enter the text macro.
> 
> So I cannot use `tapKeySeq`? Or is there some other way?

So let's backtrack a bit:
- Agent accepts *character*. 
- In regular actions, Agent translates *character* to scancode according to enUS mapping.
- UHK interprets *characters* in macro commands. 
- UHK produces *scancodes*. (If it produces *scancodes* based on *characters* in macros, it translates them based on enUS mapping.)
- OS transforms *scancodes* back to characters according to its current language keymap.

I.e., `ß` is not a scancode. It is a character, produce by our OS language (key)map. It is definitely not a legal input in any of Agent fields. 

If you want to trigger the key which has `-` label in enUS layout, you need to enter the `-` *character* into Agent's input field, which means that you need to physically tap such key combination which results in the `-` within your actual language map. On the other hand, if you are trying to make UHK produce `-` character, you first have to figure out which keys on your OS language keymap produce that character, translate them to enUS layout keys, and then enter these as *characters* into the Agent's fields.

kareltucek, 2022-06-29:
This can be shortened to:

```
$ifCtrl final tapKey b
tapKey LA-np7
```

> §testÖ 0Ö 0Ö Label not foundÖ default

It should work if you remove the dollar in `$default:`. 

Sorry about this, the firmware is somewhat in process of transition from text action with dollar notation, to command action without dollar notation... current rule is actually that  a text action is interpretted as command action if the first character of the action is `$`.

> This might make problems because of the enUS `:` which is a `Ö` on my layout. But as stated above, I cannot enter special characters in the agent...

You don't need to. All you need to type into macros or into Agent to produce any behaviour are regular ascii characters.

provinzio, 2022-06-29:
You enlightened my day. Thank you!

kareltucek, 2022-06-29:
You are welcome!