kareltucek/firmware #104: Left shift from macro not recognised as a shift in another macro

tunney, 2022-08-03:
[image]

I have the above macro, which works fine, and when I double tap left shift I get caps, and if I just press/hold it it acts as shift

[image]

I also have the above macro, which also on its own works fine, and it records a runtime macro if I press shift and a key.

The problem is with both enabled I cannot get the runtime macro to record as the shift condition is not triggered.

What have I done wrong please?

kareltucek, 2022-08-03:
For your usecase, you should use `MODMASK` notation with `input` composite mode - i.e., `$holdKey iLS-` in your shift macro.

I hope following excerpt makes the meaning clearer:

>  - `MODMASK` meaning:
>    - `{S|C|A|G}` - Shift Control Alt Gui. (Windows, Super and Gui are the same thing. ) 
>    - `[L|R]` - Left Right (which hand side modifier should be used)
>    - `{s|i|o}` - modifiers (ctrl, alt, shift, gui) exist in three composition modes within UHK - sticky, input, output:
>        - sticky modifiers are modifiers of composite shortcuts. These are applied only until next key press. In certain contexts, they >will take effect even after their activation key was released (e.g., to support alt + tab on non-base layers).
>        - input modifiers are queried by `ifMod` conditions, and can be suppressed by `suppressMods`.
>        - output modifiers are ignored by `ifMod` conditions, and are not suppressed by `suppressMods`.
>        By default:
>        - modifiers of normal non-macro scancode actions are treated as `sticky` when accompanied by a scancode.
>        - normal non-macro modifiers (not accompanied by a scancode) are treated as `input` by default.
>        - macro modifiers are treated as `output`.
>    - `{p|r|h|t}` - press release hold tap - by default corresponds to the command used to invoke the sequence, but can be overriden for any.

tunney, 2022-08-03:
Works a charm, thanks, I had not picked up on that

closed 2022-08-03 by tunney