kareltucek/firmware #105: Outputing register value to the LED display.

bistabil, 2022-08-08:
Today, I've discovered it's possible to write to the LED display and I've started playing with it.
Is it possible to output the value of a register to it?
Registers are holding numeric values while setLedTxt expects custom text.
I went through the reference manual but didn't find a command for this.

My use case:
I would like to implement a countdown timer and display its value on the LED display.

bistabil, 2022-08-08:
Even if I would go that distance and write 1000 if statements (that would require multiple macros calling each other because the number of actions is limited) I can see a problem with the performance. Evaluating 1000 statements every second let's say might be a challenge (assuming).
It would be awesome having the ability to control individual displays as that would unlock even more possibilities.

kareltucek, 2022-08-08:
No, not possible at the moment.

kareltucek, 2022-11-07:
I don't have strong feelings about the LED display. I would prefer not to flood the codebase with extensive LED display API which would pollute the namespaces and add a lot of new code which would have to be maintained.

I guess I am open to merging such functionality, although I don't intend to invest too much time into it at the moment.

If I may suggest a way forward, then please do reformulate the proposals in terms of proposed macro syntaxes, and do so in one thread for all the proposed features...

kareltucek, 2023-09-01:
Is/will be solved by https://github.com/UltimateHackingKeyboard/firmware/pull/674

closed 2023-09-01 by kareltucek