kareltucek/firmware #113: Can't save configuration to keyboard

alexthewilde, 2022-11-14:
As soon as I flash the keyboard with latest firmware fork (`v8.10.12.kt.4`), the uhk agent (`v2.0.1`) displays this message:

[image]

Saving configuration to the device is prevented with a prompt to upgrade to the latest official firmware.

The only workaround that allows me to save new configuration using the new firmware:
1. Make changes
2. Upgrade to latest official firmware
3. Flash with firmware fork
4. Save to keyboard

...for every change. So that's cumbersome.

Is this a bug or intentional by design?

alexthewilde, 2022-11-14:
@kareltucek works like a charm now. Thanks for your excellent work!

RedDirtDog, 2022-11-16:
Thanks @alexthewilde for raising this - I hit this a few days ago but didn't get around to sorting it out...

Thanks ofc to @kareltucek for all the amazing work on this. I'd been using a (very) old version, and all my macros had "write text" actions with $ commands in them, so when I upgraded the firmware, everything not only stopped working, but a lot of my keypresses would just dump out a string of commands!!

I added the `set macroEngine.extendedCommands 1` to the (newly created) $onInit macro and that seems to have worked. Not sure if there's a better way to do this, but thought I'd raise it here in case
a) a better way exists
b) someone else with old macros upgrades then wonders "where do I put this mysterious set command" lol

IzK666, 2022-11-16:
I haven't upgraded yet to the newer version, but I'll do soon (I want to make use of the led commands).

I have some doubts about the upgrade.
1) As I understand, the syntax for the macros is exactly the same that we were using with your fork, without the preceding $ symbol and, instead of write action we've to use the command action.
2) Also, if I'm not wrong, in the command field I can write multiple commands in different lines, instead of one each time.
3) And last, once I upgrade the keyboard, can I downgrade to the current version I'm using (8.10.10) and then load an old backup?

kareltucek, 2022-11-16:
@IzK666 Yes to all of them. 

(To be precise, you need the mentioned  `set macroEngine.extendedCommands 1` as mentioned above.)

IzK666, 2022-11-17:
I've created a converter from the old format to the new which can be accessed [here](https://izk666.github.io/UHK-Viewer-v5/).
It performs 3 steps: Adds the label for each layer (base, fn, mod and mouse), converts the macros from the old write action to the command, chaining them if necessary and finally creates the $onInit macro.

I think many people may find it interesting.

closed 2023-07-09 by kareltucek