kareltucek/firmware #115: calling macros that don't exist triggers other macros unexpectedly

opensensesolutions, 2023-03-14:
I just tried some command/smart macro commands for the first time with UHK.Agent-2.1.1-linux-x86_64.AppImage on ubuntu 22.04 and updated the firmware.   I delete the .config/uhk-agent directory after every uhk-agent run to avoid storing my password macros in plain text so there shouldn't be any old cruft affecting things.

If you create a command/smart macro like this:
set macroEngine.extendedCommands 1
write gonna do something weird
call nonexistantmacro

and trigger it, you get:
gonna do something weird
followed by text output by other macros that are stored in the keyboard.  for example some of the output includes:
***
://ultimatehackingkeyboard.com
iddqd

from one of the macros included by default.  It does seem repeatable, it does the same sequence of events each time, but I don't immediately see a pattern for which macros it is running, text from about a dozen different macros was output out of the several dozen that I have, it certainly isn't running them all.

I don't know if this is the expected behavior at this beta stage of development but it seems a bit dangerous, I would expect that calling a nonexistent macro would do nothing.

Also, what is the recommended way to write text with a newline after it?  \n doesn't work, I tried quoting across multiple lines, etc.  Creating a regular macro for the enter key and calling that works, and that is a fine solution, but I was just wondering if there is a better way.

opensensesolutions, 2023-03-14:
I did just notice that calling a non-existent macro does print ERR on the display, so at least a problem is flagged.  It also seemed to put the keyboard into a bad state where some other smart macros didn't work and I had to power cycle the keyboard to get things back to normal, re-saving the configuration to the keyboard ( which resets the ERR display ) didn't help.

kareltucek, 2023-03-14:
What is your firmware version?

opensensesolutions, 2023-03-14:
9.2.1 (bundled with Agent)

I see that your last release was v9.1.4 , I thought this was the upstream source but should I be reporting this to the uhk project instead?

kareltucek, 2023-03-14:
It's fine.

Yeah, I can see the problem, will get it fixed in official firmware.

Thanks for report!

kareltucek, 2023-03-14:
> Also, what is the recommended way to write text with a newline after it? \n doesn't work, I tried quoting across multiple lines, etc. Creating a regular macro for the enter key and calling that works, and that is a fine solution, but I was just wondering if there is a better way.

Using write action and not command action. Or using "tap enter" action.

(Yes, the language should indeed be refactored to support strings as clearly recognizable tokens.)

opensensesolutions, 2023-03-14:
OK, thanks, using tapKey enter is working well instead of calling a separate enter key macro, I may have been running into this:
"Beware, the state pool is small - do not use deep call trees!"
with too many levels of calls.

Thank you very much for creating these features, the extra layers and smart macros have really made some tasks less tedious and opened up completely new possibilities.

kareltucek, 2023-04-04:
Handling of non-existent macros should be fixed now!

closed 2023-04-04 by kareltucek